freifunk-berlin/firmware #442: Lede Build - OLSR plugin library versions have been updated

pmelange, 2017-03-24:
Freifunk Berlin Hedy 1.0.0-routing-master-alpha 403c57e r3205-59508e3

The libraries referenced in /etc/config/olsr and /etc/config/olsr6 need to be updated.
olsrd_txtinfo.so.0.1 -> olsrd_txtinfo.so.1.1
olsrd_nameservice.so.0.3 -> olsrd_nameservice.so.0.4
olsrd_jsoninfo.so.0.0 -> olsrd_jsoninfo.so.1.1

booo, 2017-06-25:
@SvenRoederer do you have fix for this somewhere or is this fixed upstream in a newer version of the olsrd package?

bobster-galore, 2017-09-12:
Since olsrd 0.9.0.3 will be in Hedy because it is working, this issue is not relevant for Hedy-RC. Above mentioned thingies come with olsrd 0.9.0.6.

bobster-galore, 2017-10-12:
it should be Hedy-1.1.0 or later, when new olsr-Version is used.

SvenRoederer, 2017-10-19:
:+1:  for staying with olsrd v0.9.0.3 as long as we have no clear solution for olsr on BBB-VPN (https://github.com/OLSR/olsrd/issues/20)

bobster-galore, 2018-09-20:
Do we have progress here or will we continue to postpone?

pmelange, 2018-09-21:
fixed in https://github.com/freifunk-berlin/firmware-packages/commit/ba58e285d56f975ab6f0fad4a01a434408cacfef

closed 2018-09-21 by pmelange