freifunk-berlin/firmware #443: Lede Build - OLSR6 fails

pmelange, 2017-03-24:
Build: Freifunk Berlin Hedy 1.0.0-routing-master-alpha 403c57e r3205-59508e3

After updating the plugin versions to the correct ones, starting olsr6 fails

daemon.err olsrd[7189]: olsrd exit: -- PLUGIN LOADING FAILED --

booo, 2017-06-25:
@pmelange Can you provide us with a bit more information? Maybe run olsrd by hand from the commandline and post the error log?

pmelange, 2017-06-26:
@booo I don't have any test devices right now.  I'll try to get one in the next few days and post the results when I have them.

booo, 2017-06-26:
I think the referenced commit solves the problem.

pmelange, 2017-07-16:
The commit works with both the 1.0.0-olsrd0903-alpha branch and the SAm0815_Hedy-alpha_routing-master branch.

@SvenRoederer Do you want to merge this commit to the SAm0815_Hedy-alpha_routing-master branch?

booo, 2017-07-18:
If the commit works we should merge this into the master branch. I will do so if there is no objection till the end of the week.

booo, 2017-08-11:
Merged.

closed 2017-08-11 by booo