freifunk-berlin/firmware #444: Lede Build - OLSR status pages are not showing correct data

pmelange, 2017-03-24:
The luci status page for configured interfaces does not show the actual data.
The luci status page for the smart gateway shows nothing although the plugin definately sets up a tunnel correctly.

Running a command such as 'echo "/gateways" | nc 127.0.0.1 9090' is not nicely indented and organized like it is in 0.2.0. There is also some new information provided by this command. Perhaps that is having in influence on the luci status pages.

bobster-galore, 2017-09-12:
Did u see that in recent builds? I checked:
http://lynar-luther.olsr/cgi-bin/luci/freifunk/olsr/smartgw which is ok to me. And at
http://chris-core.olsr/cgi-bin/luci/freifunk/olsr/smartgw which is 0.3.0 and no gw is selected albeit it is working on a smartgw.

bobster-galore, 2017-10-12:
any new observations? Shall we postpone to Hedy-1.1.0

SvenRoederer, 2017-11-06:
in 1b45bfce I added a patch to switch back to olsrd 0.9.0.3 in the openwrt-routing feed.
This patch also includes reverting the changes in LuCI regarding ETX-calculation

bobster-galore, 2018-09-20:
Is this still unsolved? Can we close this?

pmelange, 2018-09-21:
Seems to be solved

closed 2018-09-21 by pmelange