freifunk-berlin/firmware #445: Lede Build: The OLSR services are not being listed on the router's webpage

pmelange, 2017-03-24:
Hedy 1.0.0-routing-master-alpha 403c57e r3205-59508e3

On the default home page of the router, the Services menu option is missing.

This is probably related to #444

booo, 2017-07-05:
I can't reproduce this. Are you sure this is still an issue?

pmelange, 2017-07-16:
I finally got around to testing the SAm0815_Hedy-alpha_routing-master branch.  Unfortunately the luci services page is simply not working, but the services are properly listed in /tmp/etc/services.olsr and /tmp/etc/services.olsr.ipv6

I can also confirm that on the 1.0.0-olsrd0903-alpha branch is working just fine in luci and also in the /tmp/etc/ directory.

SvenRoederer, 2017-07-16:
```SAm0815_Hedy-alpha_routing-master``` is using >olsrd-0.9.5.x - probably the luci-page needs an adaption to the changes of the olsrd-nameservice plugin. (I assume the services coming from this plugin)

pmelange, 2017-07-16:
No, it was a general problem with luci.  Non-olsr related pages were also not working.  It might be like you said, but I can't confirm it yet.

SvenRoederer, 2017-07-16:
so LuCI was "broken" completly? which is something different than this issues headline indicates

pmelange, 2017-07-16:
Not completely broken.  Some pages.  For example /cgi-bin/luci is broken, but /cgi-bin/luci/admin is not. Everything works in admin mode.  I'm looking into it.

Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: /usr/lib/lua/luci/dispatcher.lua:380: Failed to execute function dispatcher target for entry '/'.
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: The called action terminated with an exception:
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: /usr/lib/lua/luci/dispatcher.lua:380: Failed to execute function dispatcher target for entry '/freifunk'.
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: The called action terminated with an exception:
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: /usr/lib/lua/luci/dispatcher.lua:380: Failed to execute template dispatcher target for entry '/freifunk/index'.
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: The called action terminated with an exception:
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'freifunk/index'.
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'header'.
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'themes/bootstrap/header'.
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: A runtime error occured: [string "/usr/lib/lua/luci/view/themes/bootstrap/hea..."]:150: attempt to index local 'boardinfo' (a nil value)
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: stack traceback:
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: 	[C]: in function 'assert'
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: 	/usr/lib/lua/luci/dispatcher.lua:380: in function 'dispatch'
Mon Jul 17 00:34:19 2017 daemon.err uhttpd[1703]: 	/usr/lib/lua/luci/dispatcher.lua:109: in function </usr/lib/lua/luci/dispatcher.lua:108>

bobster-galore, 2017-10-12:
it should be Hedy-1.1.0 or later, when new olsr-Version is used.

torte71, 2018-01-21:
@pmelange : This might be a problem with your build setup (see https://github.com/freifunk-berlin/firmware/issues/431 and https://github.com/openwrt/luci/issues/1521).
What is the output of "umask" on your build system? If it is different than "0022", then you should enter "umask 0022" before building the sources.

bobster-galore, 2018-09-20:
Is this still unsolved? Can we close this?

pmelange, 2018-09-21:
Seems to be solved

closed 2018-09-21 by pmelange