freifunk-berlin/firmware #446: Lede Build: https is not included

pmelange, 2017-03-24:
Hedy 1.0.0-routing-master-alpha 403c57e r3205-59508e3

- The config file for uhttp is different from 0.2.0.
- The necessary packages are not included (uhttpd-mod-tls, px5g, maybe more)

I would also recommend that redirect_https be enabled by default.

SvenRoederer, 2017-03-26:
see https://github.com/freifunk-berlin/firmware/issues/366 for redirecting

pmelange, 2017-03-26:
Go to sleep. :)

I just rebuilt the raid on the house server.  Good night.

P

On 26.03.2017 03:52, Sven Roederer wrote:
> see https://github.com/freifunk-berlin/firmware/issues/366 for redirecting
>

booo, 2017-07-05:
I think we have to add the libustream-ssl package. The init script does not generate the certs/keys for the uhttpd server because the shared library is missing:

https://github.com/lede-project/source/blob/master/package/network/services/uhttpd/files/uhttpd.init#L155

Our unique id fix in the uhttpd-defaults files is maybe obsolete.

booo, 2017-08-11:
HTTPS should work again. @pmelange could you test a master branch just to be sure?

closed 2017-08-11 by booo

pmelange, 2017-08-12:
Confirmed.  Thank you.