freifunk-berlin/firmware #447: Firewall: multiple freifunk forwardings

everloop2, 2017-03-25:
seen at branch: master | 0.3.0-rc1 | SAm0815_Hedy-alpha_olsr0903

no negative effects so far, only looks weird at LuCI > Network > Firewall

etc/config/firewall

3times:
```
config forwarding
	option dest 'freifunk'
	option src 'freifunk'
```
2times:
```
config forwarding
	option dest 'freifunk'
	option src 'lan'
```

pmelange, 2017-03-25:
Hedy 1.0.0-routing-master-alpha 403c57e r3205-59508e3

I see it here too.  But it should also be noted that there is no 'lan' firewall zone.   It has no effect as no entries are created in iptables for zone 'lan'.

Unfortunately there are multiple iptalbes entries for the 'freifunk' zone in iptables.
<pre><code>
Chain zone_freifunk_forward (5 references)
target     prot opt source               destination         
forwarding_freifunk_rule  all  --  anywhere             anywhere             /* !fw3: user chain for forwarding */
zone_freifunk_dest_ACCEPT  all  --  anywhere             anywhere             /* !fw3: forwarding freifunk -> freifunk */
zone_freifunk_dest_ACCEPT  all  --  anywhere             anywhere             /* !fw3: forwarding freifunk -> freifunk */
zone_freifunk_dest_ACCEPT  all  --  anywhere             anywhere             /* !fw3: forwarding freifunk -> freifunk */
ACCEPT     all  --  anywhere             anywhere             ctstate DNAT /* !fw3: Accept port forwards */
zone_freifunk_dest_REJECT  all  --  anywhere             anywhere             /* !fw3 */
</code></pre>

In my /etc/config/freifunk I have 
3x friefunk->freifunk
2x lan->freifunk
1x wan->freifunk.

Many of the rules are also doubled
3x tcp port 80, 442, and 22
2x tcp port 698, 17990
2x udp any port

SvenRoederer, 2017-09-24:
`/etc/config/firewall` looks good after bootup, but double entries are present after running ffwizard

bobster-galore, 2018-09-21:
Would somebody like to solve this?

pmelange, 2018-10-14:
This is still a issue with 1.1.0-alpha 4155686