freifunk-berlin/firmware #448: GL-MT300A image fails to load and reset

niccokunzmann, 2017-03-25:
I was just flashing the master image from here:

> https://util.berlin.freifunk.net/hardware?name=gl-mt300a  
> master (162) vom 2017-03-24 00:58:19

The router does not open WIFI or respond on the LAN ports.

Problem: I do not want others to flash the image and make their router unusable for some time.
Possible Solution: Can we somehow remove the image so other people do not flash the router with it?

SvenRoederer, 2017-03-26:
can you check with a original LEDE-17.01 image for working netconfig?

For now I just disabled the MT300a in master-branch

niccokunzmann, 2017-03-26:
I will try to. I will see how I can access the router - I think it is possible with a Raspberry Pi and jumper cables to the serial port.

SvenRoederer, 2017-03-26:
no success with safe-mode?

niccokunzmann, 2017-04-01:
This is the only reply I get:
[image]
And a port scan reveals this:

```
>25 - smtp<
>110 - pop3<
>119 - nntp<
>143 - imap<
>465 - <
>563 - nntps<
>587 - <
>993 - imaps<
>995 - pop3s<
```

SvenRoederer, 2017-04-04:
that's quite unrealistic. Are you connected to the right address?
None of these services is installed on the routers by default. And esp. in failsafe-mode they are not enabled in case they would have been installed.

niccokunzmann, 2017-04-16:
The image works. I do not know now what I did. I requested an IP-Address at the WAN port so I could gain access again. Closing it.

closed 2017-04-16 by niccokunzmann