freifunk-berlin/firmware #450: Ramips-mt7620 fails to build with LEDE after 0a3088cb4

SvenRoederer, 2017-03-28:
package "kmod-mt76" is missing.

in https://git.lede-project.org/?p=source.git;a=commit;h=0a3088cb4b21458dff1b5c1837ce635ec4154b5a this package was split-up and change to a meta-package. 
This metapackage will no be build and breaks some deps.
See http://lists.infradead.org/pipermail/lede-dev/2017-March/006900.html for follow up

everloop2, 2017-04-06:
Seems to be workin - did some test builds and its building now :D
Don't have hardware to test images.

branch: SAm0815_Hedy-alpha_olsr0903_fix-kmod-mt76

feeds.conf
```
src-git packages https://github.com/openwrt/packages.git^3c36d4388cfdb405d423bb0d885f693341484b55
src-git luci https://github.com/openwrt/luci.git^b47a21fbbb8c52ff54e99cb1fc68b11c04070d0d
src-git routing https://github.com/SvenRoederer/openwrt-routing.git;lede-17.01_olsrd0903
src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git;SAm0815/integrate
```

SvenRoederer, 2017-04-06:
was fixed by https://git.lede-project.org/?p=source.git;a=commitdiff;h=7099bb19b50d74d3a400aa3a60e0b7ba0157bc1e yesterday, after discussing with @nbd168

closed 2017-04-06 by SvenRoederer