freifunk-berlin/firmware #454: Lede - build fails for gpsd on ArchLinux

cholin, 2017-04-19:
Current master fails to build because of missing `-lncurses` flag for gpsd package on ArchLinux. See https://github.com/openwrt/packages/issues/3710 and the fix in https://github.com/openwrt/packages/commit/9efeb2915585739f62b38c1f86a128a9fb36b403 . Unfortunately this fix hasn't been backported to lede packages 17.01 yet (master:  https://github.com/openwrt/packages/commit/56870f42ece3f46c5000192376120aa467aa8707 )

SvenRoederer, 2017-04-19:
* gpsd is not part of out default packages-list
* as you found this small patch is fixing it, I think it's best up report this upstream (openwrt-packages or even gpsd) for a global fix

cholin, 2017-04-19:
It's already reported upstream (since january but nothing is happening there) and even that gpsd is not part of your default packages list, it gets build anyway with firmware master (actually as well if you remove alfred and gps flags).

glycoknob, 2017-05-08:
it's fixed now: https://github.com/openwrt/packages/commit/d99ed231bec135a02006a454503b007dc7f11c82

closed 2017-05-08 by cholin