freifunk-berlin/firmware #457: build 245: no vpn internet access possible on wdr4300

lightcatcher, 2017-05-06:
Hey!
using wdr4300-v1 with 243-freifunk-berlin-1.0.0-olsrd0903-alpha-tl-wdr4300-v1-sysupgrade.bin
used wizard for config(+vpn). wdr4300 is connected via lan to a fritzbox.
no ping/internet access possible via (lan+freifunk wlan)

log is attached.
if u need further information let me know.

[wdr4300-build-243-freifunk-berlin-1.0.0-olsrd0903-alpha-tl-wdr4300-v1.txt](https://github.com/freifunk-berlin/firmware/files/981088/wdr4300-build-243-freifunk-berlin-1.0.0-olsrd0903-alpha-tl-wdr4300-v1-sysupgrade.bin.txt)

SvenRoederer, 2017-05-08:
it'S this build?
``` 
https://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/243/

Firmware: git branch "SAm0815_Hedy-alpha_olsr0903", revision 0d3a4c6
OpenWRT: repository from https://git.lede-project.org/source.git, git branch "48461b5abc382b26edd0d2637f0f7b952bed90ba", revision v17.01.1
Feed packages: repository from https://github.com/openwrt/packages.git, git branch "710965eb1244af941c209da65504b09d9d5c0b80", revision 710965e
Feed luci: repository from https://github.com/openwrt/luci.git, git branch "63d61d726cd3487820f2c7ba70a2470a3c196456", revision 63d61d7
Feed routing: repository from https://github.com/SvenRoederer/openwrt-routing.git, git branch "lede-17.01_olsrd0903", revision 7e66fc8
Feed packages_berlin: repository from https://github.com/freifunk-berlin/firmware-packages.git, git branch "integrate", revision eb3bc18```

Have you seen this in other builds too (I don't remember from personal mails anymore) ?
But you mentioned this can be also seen on ERX (ramips-mt7621), correct?

lightcatcher, 2017-05-08:
yes, correct. it is also present on previous builds based on LEDE.
And i also have problems with an ubnt-erx. tried different builds, last one was (184) still not working, i can get more logs if its needed.

lightcatcher, 2017-05-12:
about the issue with the erx-device. i just installed build SAm0815_Hedy-alpha_olsr0903 (195) vom 2017-05-06 03:43:36
setup everything and its working fine.

the wdr is currently running on kathleen 0.3.0 should i test the latest build for this device ?

SvenRoederer, 2017-05-13:
build 195 is commit 0d3a4c6; build 184 was 5b2a6c3 (sadly rebased and gone)

but the only differences are 95bacd9fc2870c1e82f4193f944626a9cc226094 & d26583b6a3550bd72dbe13c040ff750c0e443e20

Which VPN-instance are you using? (as it's round-robin, you get one random of vpn03a .. vpn03f)
Can you hardcode one by one in `/etc/config/openvpn` (replace vpn03.berlin.freifunk.net by VPN03a.berlin... to vpn03f.berlin...) and reboot

lightcatcher, 2017-05-14:
at the moment, i entered vpn03e.* as main and vpn03f.* as backup and everything is working fine...
[erx-vpn-hardcoded-working.txt](https://github.com/freifunk-berlin/firmware/files/999712/erx-vpn-hardcoded-working.txt)

SvenRoederer, 2017-05-15:
can you also try the other way around? To check if one of the other instances always causing the failure?

SvenRoederer, 2017-10-02:
@lightcatcher any news on this?
Or can we just close?

closed 2017-10-02 by lightcatcher