freifunk-berlin/firmware #464: missing tcpdump package

booo, 2017-06-02:
In the default image from `Firmware Berlin (Hedy 1.0.0-olsrd0903-alpha rev e0debcd) Generic - ar71xx/generic` the tcpdump package is missing. We should check for tcpdump prior to a new release. 

Maybe we should write a small script that checks the images for packages?

SvenRoederer, 2017-06-02:
tcpdump seems to have been never included in the default image ever.

booo, 2017-06-08:
I think we should include tcpdump into the default image of all >4MB boards because it's one of the standard tools for debugging.

SvenRoederer, 2017-06-09:
* there is no user-request till now
* as there seems no real demand, it might just blow up the final images
* it can be installed anytime via opkg
* it makes it more "inviting" to sniff users data 

But feel free, if you really like it included

bobster-galore, 2017-06-13:
Since u can easily install tcpdump by opkg install tcpdump, I think we should not bloat standard images. Common users probably don't need that at all.

booo, 2017-06-14:
Well they also don't need tmux, iperf, ethtool and other stuff. Tcpdump is one of the standard tools for network debugging. It qualifies for the standard tool set on the router.

I will check how much space is left on 8MB boards and add the package to the default set if there is still enough space for other stuff afterwards.

pmelange, 2017-06-16:
tcpdump-mini is also one of the first things I install on a new router.  It would be _nice_ if it were there by default.

booo, 2017-06-16:
And by the way we can't install stuff if we are not already connected...

booo, 2017-06-18:
Situtation with build-in tcpdump:

```
root@LEDE:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 4.5M      4.5M         0 100% /rom
tmpfs                    61.5M    244.0K     61.3M   0% /tmp
/dev/mtdblock3            2.3M    296.0K      2.0M  13% /overlay
overlayfs:/overlay        2.3M    296.0K      2.0M  13% /
tmpfs                   512.0K         0    512.0K   0% /dev
```

We could install tcpdump-mini instead. The current backbone image uses tcpdump. Maybe it's a good idea to install the same package?

Anyway I think we have enough space left on 8MB devices to install one version of tcpdump.

SvenRoederer, 2017-06-25:
solved by merging PR #468

closed 2017-06-25 by SvenRoederer