freifunk-berlin/firmware #465: option ip_conntrack_max does not set conntrack_max

booo, 2017-06-08:
On  `Firmware Berlin (Hedy 1.0.0-olsrd0903-alpha rev 0d3a4c6) Generic - ar71xx/generic` aka. Schwalbenweg18-core.olsr the firewall configuration uses the option `ip_conntrack_max`:

```
config advanced                   
        option tcp_westwood '1'
        option tcp_ecn '0'   
        option ip_conntrack_max '8192'
```

Output from `sysctl -a | grep conntrack`:
```
net.netfilter.nf_conntrack_max = 16384
[...]
net.nf_conntrack_max = 16384
```
Seems like the option does not work. Current workaround is using /etc/sysctl.conf.

SvenRoederer, 2017-06-08:
Please report Upstream and fix there
package: luci-mod-admin-full

SvenRoederer, 2017-06-08:
forget about my last comment.

checking the recent sources (https://git.lede-project.org/?p=project/firewall3.git;a=summary, https://git.lede-project.org/?p=source.git;a=summary) there is no reference to ```option ip_conntrack_max``` or any other of section ```config advanced```.

In https://git.lede-project.org/?p=project/firewall3.git;a=commit;h=99499fdbe5221847288a6d18edf1032d2702cff9 the ```tcp_westwood``` was removed.
I assume the whole "advanced" section is deprecated.

I'm not sure how @pmelange made this configs and got these options from

SvenRoederer, 2017-06-08:
to me it looks more that out firewall defaults are outdated ...
https://github.com/freifunk-berlin/firmware-packages/search?utf8=%E2%9C%93&q=tcp_ecn&type=

booo, 2017-06-08:
Yes, seems like we should update our firewall defaults.

I still consider this a bug because we want to set some options and actually don't set them.

pmelange, 2017-06-16:
I didn't add the firewall optional manually.  It must have been automagically done.  I agree, the automagical firewall is a mess.  See issue #447.

booo, 2017-06-18:
Seems like the section is now called `defaults`: https://lede-project.org/docs/user-guide/firewall_configuration

But I can't find a conntrack_max option.

I will rename the section and remove the conntrack_max option. I think ~16k is a good standard value. Let us consider a increase of the default value later on in the process.

booo, 2017-07-06:
We enable `tcp_westwood` in our configuration. Does this make sense after all? 

The `syn_flood` option was renamed to `synflood_protect`. We set the default value explicitly. I think we can remove this and work with the default value.

bobster-galore, 2017-10-04:
As @SvenRoederer mentioned westwood was removed 2013, so no need for settings.
Here is the default values of LEDE for conntrack: https://github.com/lede-project/source/blob/7765e442d04e4c19690f81084a9726776aea8b76/package/base-files/files/etc/sysctl.conf

This is may be ok with us or?

bobster-galore, 2017-10-04:
> Seems like the section is now called defaults: https://lede-project.org/docs/user-guide/firewall_configuration

But I can't find a conntrack_max option.

It's almost there, search for conntrack instead of conntrack_max as it is only an option of conntrack. s.a. https://lede-project.org/docs/user-guide/firewall_configuration#notes_on_connection_tracking

SvenRoederer, 2017-10-09:
To me it looks like the whole firewalling needs an update.
But as there seems to be no real bug, we should have a look into this in an other moment and concentrate on release-blocking issues for now. Which this itn't, so I'll remove the "Hedy-1.0.0" milestone.

bobster-galore, 2018-09-21:
Would somebody like to solve this?

bobster-galore, 2018-09-22:
@booo where did u want to set it?
The actual configuration even doesn't show a network/firewall in luci.
Now it's set in /etc/sysctl.conf and working.
Some additional information on conntrack_max is found [here](http://antmeetspenguin.blogspot.com/2011/01/high-performance-linux-router.html)
Can we track down a solution for the conntrack-stuff and close this?