freifunk-berlin/firmware #466: 802.11s setting individual mesh_id not possible on Hedy alpha

bobster-galore, 2017-06-10:
Hej,
it is not possible to use an individual mesh_id. Independent of the settings in 
/etc/config/wireless
...
config wifi-iface
        option device 'radio0'
        option ssid 'smesh'
        option mesh_id 'smesh'
        option encryption 'none'
        option mode 'mesh'
        option network 'mesh'
        option key ''
...
it's always "openwrt" on air.
It seems to be somewhere in the driver since there is no string like that in the configuration files.
Any hints?
cheerio bob

SvenRoederer, 2017-06-10:
can you check with a current LEDE snapshot build?

bobster-galore, 2017-06-11:
Oooh ooh smesh it up, smesh it up, smesh it up
to go with the damned, LEDE snapshot worx!
latest Freifunk build Freifunk Berlin Hedy 1.0.0-olsrd0903-alpha 4d4151f r2993+442-b9a408c / LuCI lede-17.01 branch (git-17.152.82987-7f6fc16) worx as well, only Luci doesn*t show the mesh_id.

booo, 2017-06-20:
Please report the problem with luci upstream: https://github.com/openwrt/luci

I'm closing this because it's a upstream problem.

closed 2017-06-20 by booo

SvenRoederer, 2017-06-21:
It would be great to habe a reference to the upstream ticket Herr.

bobster-galore, 2017-07-04:
https://github.com/openwrt/luci/issues/1228

SvenRoederer, 2017-09-13:
@bobster-galore can you test with a recent SAm0815_Hedy-alpha* build?