freifunk-berlin/firmware #467: Support for TP-Link CPE210 V1.1EU

Proxycon, 2017-06-11:
Request to a firmware supporting the TP-Link CPE210 V1.1**EU**. If I try to flash the build for V1 / V1.1 I get a "hardware not supported error".

SvenRoederer, 2017-06-11:
which FW-version have you used?
have you read https://wiki.freifunk.net/Berlin:Firmware#cite_note-FCC_regionlock-5 ?

Proxycon, 2017-06-18:
I used the berlin community firmware [kathleen-0.3.0-cpe210-220-510-520-factory](https://util.berlin.freifunk.net/hardware?name=cpe210-220,cpe210-220-510-520).
Does the cite note mean it only works with the original firmware and is not supported? If thats the case i would like to request a version for the V1.1EU. Thats why i started this issue in the first place.

SvenRoederer, 2017-06-18:
You have to use the current development firmware "Hedy". This received upstream support for the EU-models.
It's also linked on the firmware-downloadpage.

But I strongly recommend you to use build "SAm0815_Hedy-alpha_olsr0903" or "SAm0815_Hedy-alpha_routing-master" as the regular master-build has some known issues
- https://github.com/freifunk-berlin/firmware/issues/435
- https://github.com/freifunk-berlin/firmware/issues/427
- https://github.com/freifunk-berlin/firmware/issues/422

SvenRoederer, 2017-06-25:
@Proxycon is your issue solved?

Proxycon, 2017-06-25:
Not quite but that has little to do with the software than with me not knowing exactly how to get a uploadable file of the brach. I tried downloading the hole branch from github but it doesn't looked liked there was a single file with the purpose of a flash. How / Where do I find a file? Do I have to use the buildbot?

SvenRoederer, 2017-06-25:
follow:
* https://wiki.freifunk.net/Berlin:Firmware#WLAN-Router
* in the CPE210 line go to column "Firmware"
* on the linked page select "Auch alte Releases und Development-Branches anzeigen"
* the following page shows extended downloads
* on the lower area you find the "Development" subsection, with the branches I referenced above

Also have in mind, that you have to shorten the filenames, that they will be accepted by the TPlink Web-updater

Proxycon, 2017-06-25:
Thanks a lot for your kind help. I will come back if I encounter issues.

SvenRoederer, 2017-06-25:
So let's close this for now.
Feel free to reopen if you encounter issues.

closed 2017-06-25 by SvenRoederer