freifunk-berlin/firmware #470: Failing to resolve vpn03

mauricemeyer, 2017-07-01:
Hey everyone,

I just set up a Ubiquiti Unifi AP with Firware version 0.3.0 yesterday. Following the wizard, configuration was easy and nothing unexpected happened. Following the reboot, the connection to VPN03 fails.

I used the "Connect and share my Internet" option in the configuration wizard, the OpenVPN config also looks good to me

The apparent problem is that the name can't be resolved:

```
Sat Jul  1 14:44:01 2017 daemon.err openvpn(ffvpn)[2980]: RESOLVE: Cannot resolve host address: vpn03.berlin.freifunk.net: Name or service not known
Sat Jul  1 14:44:01 2017 daemon.err openvpn(ffvpn)[2980]: RESOLVE: Cannot resolve host address: vpn03.berlin.freifunk.net: Name or service not known
Sat Jul  1 14:44:01 2017 daemon.notice openvpn(ffvpn)[2980]: SIGUSR1[soft,init_instance] received, process restarting
Sat Jul  1 14:44:03 2017 daemon.warn openvpn(ffvpn)[2980]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sat Jul  1 14:44:03 2017 daemon.err openvpn(ffvpn)[2980]: RESOLVE: Cannot resolve host address: vpn03-backup.berlin.freifunk.net: Name or service not known
Sat Jul  1 14:44:03 2017 daemon.err openvpn(ffvpn)[2980]: RESOLVE: Cannot resolve host address: vpn03-backup.berlin.freifunk.net: Name or service not known
Sat Jul  1 14:44:03 2017 daemon.notice openvpn(ffvpn)[2980]: SIGUSR1[soft,init_instance] received, process restarting
```

Digging a bit in the config, I tried to get a connection to any of the nameservers listed in `/tmp/resolv.conf.auto`:

```
nameserver 85.214.20.141
nameserver 213.73.91.35
nameserver 194.150.168.168
nameserver 2001:4ce8::53
nameserver 2001:910:800::12
```

The 3 IPv4 nameservers can't be reached because none of their networks is reachable, the two IPv6 nameservers can't at least be pinged because of a `ping: sendto: Permission denied` error, but I didn't find anything related to that in the firewall config.

I see that the router is not having an IP on `eth0` although my DHCP server allocated one for it, I guess that's the issue most relevant here.

Any hints/help on how to fix this?

PolynomialDivision, 2017-07-02:
Did you try to give `eth0` a static IP and enter the gateway manually?

You can do this by editing `/etc/config/network`.

    config interface 'et0'
    	option netmask '255.255.255.255'
    	option proto 'static'
    	option ipaddr '192.168.1.2'
        option gateway '192.168.1.1'

And then `/etc/init.d/network reload`.

Maybe you should look at `/etc/config/firewall` and `/etc/config/dhcp`.

Another hint would be to look at the traffic on eth0.

mauricemeyer, 2017-07-02:
@PolynomialDivision thanks for the hint to `/etc/config/network`.

Unfortunately, making the ip address on `eth0` static didn't change anything as reloading or restarting the network service didn't assign it to the interface. (even rebooting didn't help)

firewall and dhcp config aren't altered and look good to me.

There is absolutely no traffic on `eth0` as it just has an ULA address. Maybe I'm looking at the wrong interface? If that would help further, I could dump all/any needed config/addresses etc.?

sarumpaet, 2017-07-04:
Possibly this is another instance of https://wiki.freifunk.net/Berlin:Firmware#cite_note-lanports-1 / #292 ?

PolynomialDivision, 2017-07-04:
> Unfortunately, making the ip address on eth0 static didn't change anything as reloading or restarting the network service didn't assign it to the interface. (even rebooting didn't help)

Maybe you can try to use ifconfig for assigning...

    ifconfig eth0 192.168.1.2 netmask 255.255.255.0 up

SvenRoederer, 2017-07-04:
it looks like the uci.config provided by @PolynomialDivision is wrong. 
```
config interface 'wan'
	option netmask '255.255.255.0'
	option proto 'static'
	option ipaddr '192.168.1.2'
        option gateway '192.168.1.1'
```

mauricemeyer, 2017-07-18:
Thanks @SvenRoederer and @PolynomialDivision, I could fix the problem as follows:

I changed the `option ifname` of the `dhcp` interface to `eth1` (was `eth0`) and set the `ifname` of `wan` and `wan6` to `eth0` (was `eth1`). Then I restarted the network and everything works fine.

I didn't set the `wan` interface configuration static as it works fine with dhcp now.

Is this a permanent solution or will future firmware upgrades reset those settings?

PolynomialDivision, 2017-07-18:
Sorry I have no Ubiquiti Unifi AP.
This sound like you mixed up your wan and lan socket. ^^
Are u sure that the cables are connected the right way?

So now you just switch wan and lan sockets...
You could try to reset the config and switch the cable connection... ^^

Or you just stay with that config. When you sysupgrade via Luci: Check the "Keep settings"-option. :)

mauricemeyer, 2017-07-18:
That's what I'm wondering about, because the Unifi UAP does only have one physical interface.

bobster-galore, 2017-07-18:
We set up 2 unifi AP recently with 0.2.0, which worked out of the box. Only one needed a cold boot. I think there should be no difference to 0.3.0. weird?!

SvenRoederer, 2017-09-24:
so it seems also to be another instance of a single-LANport router using it's LAN-port as uplink and not to provide FF-LAN for clients. 
So https://github.com/freifunk-berlin/firmware/issues/292 get's another point

closed 2017-09-24 by SvenRoederer