freifunk-berlin/firmware #472: [RFC] einheitliche Netzwerk-konfig für uplinks mit / ohne Tunnel

SvenRoederer, 2017-07-04:
Wenn wir in Zukunft Setups unterstützen wollen, die direkt ausleiten, als auch Uplink-Anbindung via Netzwerktunnel sollten wir ein flexibels Netzwerkmodel haben.
* die genutzten interfaces sollten klar sein
* die Konfiguration sollte sich für die wesentlichen Teile des OS nicht unterscheiden (keine unterschiedlichen Scripte, je nach Uplink)
* per default sollte kein Zugriff auf das private LAN des Users möglich sein, aber durch den User konfigurierbar
* möglichst  Verzicht auf die Firewall; Nutzung von Policyroputing

SvenRoederer, 2017-07-04:
Ich hab dazu mal folgenden Ansatz probiert:
* das jetzige "ffvpn"-Interface wird zu "ffuplink" (dadurch kann das interface in jedem Setup sinnig genutzt werden)
* für Tunnel ist das "ffuplink" das lokale Interface des Tunnels
* für direktes Ausleiten  "bridged" das lokale ffuplink-Interface auf die WAN-schnittstelle
  * Erzeugung des lokalen ffuplink interfac via "veth"
  * ein Ende des veth-paares wird verbunden auf die WAN-schnittstelle (bridge)
  * das andere Ende steht als normales Interface zur Verfügung
  * mit "udhcpc -i ffuplink" kann eine separate IP-Adresse von uplink bezogen werden
    * der Router benötigt 2 Upstream-Ipadressen
``` 
root@Ahof-frieden03:~# ip link add ffuplink type veth peer name ffuplink_slave
root@Ahof-frieden03:~# brctl addif br-wan ffuplink_slave
root@Ahof-frieden03:~# ip link set up ffuplink_slave
root@Ahof-frieden03:~# ip link set up ffuplink
```

booo, 2017-07-04:
Könntest du erklären **warum** wir das so bauen sollten?

SvenRoederer, 2017-07-04:
So können alle tools, scripte immer das gleiche setup nutzen und müssen nicht abhängig vom verwendeten Uplink mal auf "WAN", "ffvpn" oder was auch immer arbeiten.
Das betrifft u.a.:
- Policyrouting
- Trafficshaping
- Firewall
- Monitoring

So kann jede Art von Uplink als "plugin" ins System gehangen werden, ohne dass man an jedes script ran muss für "If ..  then"-orgien

pmelange, 2017-07-16:
I disagree with WAN needing an ip address.  It can be completely unmanaged and the private network should function smoothly.  Maybe we shouldn't call it WAN anyhow.  Perhaps private-lan would be better.

We are going to have to NAT the traffic over the private network with the direct approach, which means a different firewall script than with our current openvpn setup.  But we could also decide to NAT everything before going out any tunnels too.

andibraeu, 2017-07-17:
Zum letzten Firmware Hackathon haben wir einen Ansatz erarbeitet:

[image]

Das Ganze arbeitet hauptsächlich mit Policyrouting. Firewallregeln werden lediglich benötigt, um das NAT zu ermöglichen. Es ermöglicht uns auch, mit und ohne VPN zu arbeiten.

Den privaten Teil des Netzes realisieren wir nicht mehr als Bridge, sondern als eigenes Netz auf dem Router. Das hat den Vorteil, dass auch Nutzer ohne Internetrouter ein privates Netz haben können.

Das hat @andrenarchy mal umgesetzt: https://github.com/freifunk-berlin/firmware-packages/pull/116

Eine Nuss, die wir noch knacken müssen: Das dyngw-Plugin möchte noch nicht so richtig und versucht, Internet über Routen im Main Table zu finden.

BTW: im neuen Wizard bauen wir gerade die Möglichkeit ein, beliebige OpenVPN-Server sowie MeshVPN zu konfigurieren. Außerdem ist eine sysupgrade-Seite in Arbeit. Wer mithelfen möchte, ist gern eingeladen!

pmelange, 2017-07-17:
I agree, that when a ffrouter does not connect to an already existing private network (mesh only), that it would be good to have to option to create one.  

But if the ffrouter does connect to an already existing private network, then I think the expanded private network on the ffrouter should share the same address space.  Using a separate address space breaks any type of local network services (NAS, Printer) which happen to be connected to the ffrouter's private network.

andibraeu, 2017-07-17:
We discussed that. But we decided to reduce complexity. You don't need to use this private network part of  the freifunk router if you already have one.

BTW: Seems as the bridge solution will be removed soon: https://github.com/freifunk-berlin/firmware-packages/pull/107

pmelange, 2017-07-17:
If you follow the thread, it was also mentioned that this feature is useful/desired, and that manual configuration should always be possible.  As long as the new networking architecture doesn't prevent setting up the private-lan, then I think it is acceptable.  A howto would be necessary if it's not available with the wizard/wizard2.

Still, I don't agree with a separate address space for the private-lan as showen in the photo above.  And if it's an unmanaged interface, then it shouldn't cause any trouble with dyngw or routing.

SvenRoederer, 2017-07-17:
Am 17.07.2017 um 01:16 schrieb pmelange:
> I disagree with WAN needing an ip address. It can be completely unmanaged and the private network should function smoothly. Maybe we shouldn't call it WAN anyhow. Perhaps private-lan would be better.
> 

Indeed the ipaddress of the original WAN-Interface might become useless and might not be useful in a direct-routing setup.
But my idea is to keep this WAN-address rather than moving it over to interface "ffuplink-slave". having 2 addressens will prevent us from inventing some "magic" to move the address over to the other
interface and will let the wan-interface act as normal no matter this kind of network-config.
removing the WAN-Interface at all might be a bad idea, as a lot of LEDE-functionality might directly rely on "WAN"

> We are going to have to NAT the traffic over the private network with the direct approach, which means a different firewall script than with our current openvpn setup. But we could also decide to NAT
> everything before going out any tunnels too.
> 
To NAT any outgoing traffic is the way that is supposed to go, as suggested by @booo in https://github.com/freifunk-berlin/firmware/issues/416
So it's implemented in the current "SAm0815_uplink" builds

SvenRoederer, 2017-07-17:
Am 17.07.2017 um 11:33 schrieb Andi Bräu:
> BTW: Seems as the bridge solution will be removed soon: freifunk-berlin/firmware-packages#107 <https://github.com/freifunk-berlin/firmware-packages/pull/107>

no, instead I suggest to have the WAN-Interface as a bridge by default. (see  https://github.com/freifunk-berlin/firmware-packages/commit/9cf77ec8043c8da8acd4bafa0139449dcd539081), and not only then
the "Private AP"-feature is enabled in the wizard1
(defaults/freifunk-berlin-network-defaults/uci-defaults/freifunk-berlin-network-defaults: uci set network.wan.type=bridge)

SvenRoederer, 2017-09-04:
andrenarchy hatte mir letztens ein paar ergänzende Infos zu dem Ansatz (https://github.com/freifunk-berlin/firmware/issues/472#issuecomment-315691045) gegeben.
Ich frage mich, ob es sinnvoll ist, die aktuell vonhandenen Interfaces um zu benennen. Die meisten Interfaces haben wir schon in der aktuellen Firmware mit den bakannten Namen (dhcp, wlanx-adhoc-y). 
Eine Umbenennung bei gleicher Funktion verwirrt wohl die meisten sporadischen User, die sich an die aktuellen Namen gewöhnt haben.

bobster-galore, 2017-09-08:
There are users around who are quite familiar with the existing interface names and are easy to configure them. I see no advantage in renaming existing interfaces.
Furthermore the existing names are quite descriptive s.a.
wlan0-adhoc-2 first wlan meshing on 2 GHz, why should we encrypt it to mesh1?

bobster-galore, 2017-10-04:
I think we tinkered enough about this, there is a foundation with LEDE, where we can use every kind of wizard on. The renaming of something is a different subject and can be done later or not at all. We should release Hedy 1.0.0.

SvenRoederer, 2017-11-17:
was implemented in freifunk-berlin-packages with commit https://github.com/freifunk-berlin/firmware/commit/a33755ecf32e0b91d0eea2f8be27d3577ff520da

closed 2017-11-17 by SvenRoederer