freifunk-berlin/firmware #479: wifi on after flash

bobster-galore, 2017-08-26:
wifi enabled after flash would be really nice, to have a second possibility to connect to the router for configuring. I think security concerns are neglectable because I presume routers are configured in one go. So there's not that much time to capture the router while configuration.
How u mean?

booo, 2017-08-31:
Normally you can enable the wifi by pressing the wifi button.

SvenRoederer, 2017-09-02:
I've seen routers around, which do not have a wifi-button or any other button.

bobster-galore, 2017-09-08:
3 of 5 of my routers don't have a wifi-button.
The next generation CPE-210 will have only one port so there's no wifi button and no dhcp on a port.

bobster-galore, 2017-10-08:
Is withdrawn after community-discussion https://lists.berlin.freifunk.net/pipermail/berlin/2017-October/036443.html

closed 2017-10-08 by bobster-galore