freifunk-berlin/firmware #483: ssh key backbone maintainer package

booo, 2017-08-31:
There should be a package with all the public ssh keys of maintainers of the freifunk berlin backbone. The package should append the ssh keys to an existing /etc/dropbear/authorized_keys file. The package should be optional and installed on an individual basis by the core maintainer.

pmelange, 2017-08-31:
I find this a good idea.  Possibly an addition to this package would be to have a general backbone ssh-key which would allow ssh connections between the nodes directly.

With boo's proposal, we potentially have a maintenance issue, where lots of merges will be required as more people want their public keys pushed onto the nodes.  Then all nodes in the network will have to upgrade their backbone-public-keys package.

With the backbone-allnodes-keys proposal there is the security risk that the shared private key will not be private to anyone.

andrenarchy, 2017-08-31:
+1 for the package! I don't think we need a backbone key, `ssh -A` should be enough, right?