freifunk-berlin/firmware #485: make owm.lua more verbose

bobster-galore, 2017-09-22:
I found that owm.lua will always end silent. Independent of being successful or having an error.
I suggest programming it more verbose. Like --help giving an info on --dry-run or e.g. an error message because of missing coordinates or even more.
What do you think?
taken from #435

SvenRoederer, 2017-09-23:
the success you can see on the result returned by the http-client library (Updated ok, failed, probably also timeout or "host not found").
when you check the source, you will see, that with "--dry-run" oyu get a warning when no coordinates are supplied.

As the own.lua script is so short, I don't think a special "--help" is needed, as the code can be understand very easily.

Akira25, 2020-04-10:
In case of user friendliness, most of our not-so-advanced-users will benefit from a good UI. Presumably they might not be able to read the source-code.
I think a good compromise could be PR freifunk-berlin/firmware-packages#202 If the user tries to start owm.lua manually and gets nothing on his screen, he might try `owm.lua -h` and will be delivered with a little explanatory text.

pmelange, 2020-04-12:
Fixed with https://github.com/freifunk-berlin/firmware-packages/pull/202

Closing

closed 2020-04-12 by pmelange