freifunk-berlin/firmware #486: firewall-defaults: forwarding inside "freifunk" zone

SvenRoederer, 2017-09-26:
in https://github.com/freifunk-berlin/firmware-packages/blob/master/defaults/freifunk-berlin-firewall-defaults/uci-defaults/freifunk-berlin-firewall-defaults#L86 we define a forwarding from zone `freifunk` to zone `freifunk`.
This don't make sense, as by default traffic inside a firewall-zone is allowed by default.

pmelange, 2017-10-23:
Forwarding within a zone can be set to ACCEPT, REJECT, or DROP.  But that's besides the point.  I never liked this setup.  It looks strange on the web page and it is easy to break if you change the forwarding.  It is not possible to click on anything to remake this strange forwarding.

Issue #447 is related.

bobster-galore, 2018-09-21:
Would somebody like to solve this?

pmelange, 2018-10-14:
This is still a issue with 1.1.0-alpha 4155686

bobster-galore, 2018-10-17:
Yesss, could u recommend somebody to take care of it?
Would @egmont1227 be interested?
Dere are some firewall rulez (some issues) which should be overseen and would be nice to be streamlined.