freifunk-berlin/firmware #490: sec: dnsmasq >=  Version 2.78

bobster-galore, 2017-10-30:
There have been new DNS vulnerabilities and exploits found, s.a. https://www.heise.de/security/meldung/Sicherheitsluecken-im-freien-DNS-Server-Dnsmasq-gefaehrden-IoT-Geraete-Linux-Smartphones-Co-3849403.html#mobile_detect_force_desktop , when will an updated version be on freifunk-fw?

SvenRoederer, 2017-10-30:
in 7739d59bfe07d68eb1a777d77827b85f30ff1c9b master changed to include https://github.com/lede-project/source/commit/4b4a4af814e1602f7ef12c5d07ff15c4fa2570ca (dnsmasq: bump to v2.78).

SvenRoederer, 2017-10-30:
how to fix is outlined here: https://lists.berlin.freifunk.net/pipermail/berlin/2017-October/036537.html

SvenRoederer, 2017-10-30:
for the 0.3.0-series: I don't think anyone is willing to create a security-fix release here anymore.

commit 6d79975c (2017-08-27): feeds: update berlin-packages (Aug 28, 2017)

- fix #490 (LuCI throws exception after merging packages#113)
  caused by missing uci "freifunk"
- split OpenVPN-config into generic OpenVPN-setup and VPN03-specifics
  (https://github.com/freifunk-berlin/firmware-packages/pull/110)
closed 2017-10-30 by SvenRoederer