freifunk-berlin/firmware #492: switch https for opkg package-download

SvenRoederer, 2017-11-17:
in https://github.com/freifunk-berlin/firmware/issues/148 there was a request to support https for opkg-sources. This was implemented by the switch from OpenWrt to LEDE (https://github.com/freifunk-berlin/firmware/commit/d888c21df67c5f2e424fa2955849890ac4a66be8).

We can't switch to ssl download by default, as the 4MB-devices have not enough flash to contain the SSL-lib (even mbedtls) and the ca-certrs.

SvenRoederer, 2017-11-17:
We can change the ```/etc/opkg/distfeeds.conf```` in the postinst-hook, for the >4MB routers. So we have the recent hardware doing the ssl and also support for the legacy 4MB-routers.

The downside is, that this different config might cause debug-headache ....

bobster-galore, 2018-09-21:
https would be really nice! Who would care for 4MB-devices?