freifunk-berlin/firmware #493: OpenVPN-based tunnels: start individual instance only

SvenRoederer, 2017-11-17:
in https://github.com/freifunk-berlin/firmware-packages/blob/master/defaults/freifunk-berlin-openvpn-files/openvpn/60-ffopenvpn#L30 starten / stoppen wir alle OpenVPN-instanzen, wenn wir den Tunnel für den Uplink aufbauen.
Seit LEDE-master commit https://github.com/lede-project/source/commit/e2f25e607d2092cffa45196e7997854feb464232 können einzelne Instanzen angegeben werden. 

Wir sollten hier den "ffuplink" parameter ergänzen.

SvenRoederer, 2017-12-14:
This was also ported to LEDE-17.01 in https://git.lede-project.org/?p=source.git;a=commit;h=91e48304a9279358537c939bef5e0a678cb621fc

bobster-galore, 2018-09-21:
Is this mandatory or nice to have?
Or is this ok for the next milestone?

SvenRoederer, 2019-12-17:
As our OpenVPN-Setup is deprecated, and we officially do not build images that feature OpenVPN-tunnels, it doesn't make sense to invest time here.

So I'm closing....

closed 2019-12-17 by SvenRoederer