freifunk-berlin/firmware #497: Firmware Raspberry Pi 3

kehugter, 2017-12-16:
The new Raspberry Pi config in 7251b6648f7fa1f1e1fee658333616bf273fdab7 wasn't building for me. First there seems to be a typo in `brcm2708-brcm2708.confg` and `brcm2708-brcm2708.profiles`, `SUBTARGET` should be `bcm2708`. Otherwise make was failing with

```text
ls: cannot access '/home/me/code/freifunk-berlin-firmware/lede/bin/targets/brcm2708/brcm2708/*-imagebuilder-*.tar.xz': No such file or directory
mkdir -p /home/me/code/freifunk-berlin-firmware/firmwares/brcm2708-brcm2708
./assemble_firmware.sh -p "RaspberryPi" -i  -e /home/me/code/freifunk-berlin-firmware/embedded-files -t /home/me/code/freifunk-berlin-firmware/firmwares/brcm2708-brcm2708 -u "default tunnel-berlin backbone"
IB_FILE does not exist -e
make: *** [Makefile:185: .stamp-images] Error 1
```

After changing that, `lede/target/imagebuilder/files/Makefile` complained that

```text
Profile "RaspberryPi" does not exist!
```

So I ran LEDE's `make menuconfig` to get their config parameters for the Raspi 3 and feed them to freifunk, which seems to have worked. `make -j4 TARGET=brcm2708-bcm2710` made a clean exit at least. The question is: is this image valid?

The files I used:

brcm2708-bcm2710.confg
```text
CONFIG_TARGET_brcm2708=y
CONFIG_TARGET_brcm2708_bcm2710=y
CONFIG_TARGET_brcm2708_bcm2710_Default=y
CONFIG_KERNEL_RELAY=y
CONFIG_PACKAGE_MAC80211_DEBUGFS=y
CONFIG_PACKAGE_MAC80211_MESH=y
CONFIG_PACKAGE_kmod-ath=y
CONFIG_PACKAGE_kmod-ath10k=y
CONFIG_PACKAGE_kmod-ath5k=y
CONFIG_PACKAGE_kmod-ath9k=y
CONFIG_PACKAGE_kmod-ath9k-common=y
CONFIG_PACKAGE_kmod-ath9k-htc=y
CONFIG_PACKAGE_kmod-rt2800-lib=y
CONFIG_PACKAGE_kmod-rt2800-usb=y
CONFIG_PACKAGE_kmod-rt2x00-lib=y
CONFIG_PACKAGE_kmod-rt2x00-usb=y

CONFIG_DEFAULT_kmod-rt2800-lib=y
CONFIG_DEFAULT_kmod-rt2800-usb=y
CONFIG_DEFAULT_kmod-rt2x00-lib=y
CONFIG_DEFAULT_kmod-rt2x00-usb=y
```
brcm2708-bcm2710.profiles
```text
rpi-3
```

kehugter, 2017-12-16:
Config files also available at https://github.com/kehugter/firmware/commit/622fd3f34f52c9bdb87a4d000990a88438d9433a

SvenRoederer, 2017-12-17:
* confirming the typo in the filenames
* according to ```make info``` of imagebuilder the correct profile is ```rpi```

SvenRoederer, 2017-12-17:
I don't know if all these wifi-drivers needs to be included, but the diff between bcm2708 and bcm2710 configs look fine.

@kehugter have you tested the generated images?

kehugter, 2017-12-17:
> according to `make info` of imagebuilder the correct profile is `rpi`

Isn't that only for the older/smaller Raspis? I'm basing by assumption in `lede/target/linux/brcm2708/image/Makefile`

```make
define Device/rpi
  DEVICE_TITLE := Raspberry Pi B/B+/CM/Zero/ZeroW
  DEVICE_DTS := bcm2708-rpi-b bcm2708-rpi-b-plus bcm2708-rpi-cm bcm2708-rpi-0-w
  DEVICE_PACKAGES := brcmfmac-firmware-43430-sdio kmod-brcmfmac wpad-mini
endef
ifeq ($(SUBTARGET),bcm2708)
  TARGET_DEVICES += rpi
endif

define Device/rpi-2
  DEVICE_TITLE := Raspberry Pi 2 B
  DEVICE_DTS := bcm2709-rpi-2-b
endef
ifeq ($(SUBTARGET),bcm2709)
  TARGET_DEVICES += rpi-2
endif

define Device/rpi-3
  DEVICE_TITLE := Raspberry Pi 3 B/CM
  DEVICE_DTS := bcm2710-rpi-3-b bcm2710-rpi-cm3
  DEVICE_PACKAGES := brcmfmac-firmware-43430-sdio kmod-brcmfmac wpad-mini
endef
```

> @kehugter have you tested the generated images?

No, after building I realised that I didn't have any spare SD card left. I'll buy one tomorrow.

SvenRoederer, 2017-12-17:
correct bcm2708 is only for the older / smaller units. bcm2710 is for pri-3

SvenRoederer, 2017-12-20:
rpi1 fixed by 7e68b61228c6d760a8dfd8a1db3976ff9d12fb07

SvenRoederer, 2017-12-28:
@torte71 as mentioned in issue #500 and issue #501 you seem to run on a RPi3. 
can you confirm that these configs in the repo can be used to create a running image?

torte71, 2017-12-28:
Yes, it works.
I had to change eth0 to be used for wan instead of br-dhcp**1, so I could use it as the uplink port (and disable the mesh/intern-ch-13 AP, because the driver seems to allow only one virtual AP).
And the work-around for the other two mentioned bugs were needed as well.
[**1 using it as br-dhcp was necessary to configure the device, so this works, too ]
But apart from that, the PI3 now works nicely as a tunnel-berlin**2 Freifunk router, that can be used via WiFi.
This was not a long-term test, but I plan to use the PI3 image as a Freifunk router at work (starting somewhere in january), then I can post long-term test results as well.
[**2 I didn't test the no-tunnel or vpn03 flavour ]

torte71, 2017-12-28:
And just for the records (PI related):
Changing the wifi channel doesn't work on-the-fly (even changing it via "iw dev" or "iw phy" just gives an error #95) - but after a reboot, the channel is changed to the value in /etc/config/wireless (so I think, setting it once is ok, but changing it aftwerwards fails).
And choosing channel #13 doesn't work (even though it is listed available in "iw"s output).
But I think, that's sth for the LEDE developers.
Btw., I didn't find a LEDE bugtracker yet (Do they hide in google? Am I blind?) Could you post a link, where I could dump my LEDE issues?

kehugter, 2017-12-28:
> Could you post a link, where I could dump my LEDE issues?

https://lede-project.org/bugs

SvenRoederer, 2017-12-29:
@torte71 regarding the VAP issue
I'm not sure on the WiFi-chip used in the RPi3, but e.g. on ath-10k it's known thar the support of adhoc is nearly broken.
Probably using 802.11s will help (see https://github.com/freifunk-berlin/firmware/issues/346 also)

SvenRoederer, 2017-12-29:
As the RPi3 board is reported to boot with the current config, I'll merge it into master.

SvenRoederer, 2017-12-29:
merged in 696871104f90b7e99da6d0293cd2f7035d5f20e4

closed 2017-12-29 by SvenRoederer

kehugter, 2017-12-30:
I finally flashed my build today, sorry for taking so long to test @SvenRoederer (Christmas getting in the way). It boots, as you mentioned.

The `trier.freifunk.net` SSID was visible for a moment after the second boot (ie, first boot after configuring), then disappeared. It doesn't seem to be available again on later reboots. Normal?

I'll go trough torte71's reports to get it working.

torte71, 2017-12-30:
- Disable the AdHoc network? The driver disables wifi completely, if there is more than one active virtual AP.
- Choose a channel between 1 and 11
- If you see more than one entry for "ESSID": Edit /etc/config/wireless and replace the "list ssid ..." lines by a single "option ssid ..."
- If you don't see the webinterface at all, edit /usr/lib/lua/luci/view/themes/bootstrap/header.htm and replace the line
`local boardinfo = util.ubus("system", "board")`
by
`local boardinfo = { hostname = "dummy" }`

SvenRoederer, 2017-12-30:
as the above issue are not related to the generic RPi3 support, feel free to open a new issue for these.
Even it feels to me that this is some issue in Upstream

torte71, 2017-12-31:
Hmm...
1: Multiple VAPs: Is PI specific
2: Channel 13 not working: Is PI specific
3: Multiple ESSID fields: Already here: https://github.com/freifunk-berlin/firmware/issues/500
4: luci/ubus/boardinfo: Already here: https://github.com/freifunk-berlin/firmware/issues/431

But I'll add some other issues :)