freifunk-berlin/firmware #501: SAm0815_Hedy-alpha: Deactivated wireless APs show wrong settings

torte71, 2017-12-26:
Branch: SAm0815_Hedy-alpha
(checked out via git today, 2017-12-26 15:00; built for PI3; TARGET=brcm2708-bcm2710)
Same behaviour in this image:
https://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/572/tunnel-berlin/freifunk-berlin-1.0.0-alpha-sam0815-43478e9-tl-wr842n-v3-sysupgrade.bin

If you disable one of the wireless virtual APs (e.g. wlan0-adhoc-2), it will display the settings from the other, non-disabled AP. E.g. the name will be displayed as "wlan0-dhcp-2" instead of "wlan0-adhoc-2" and the type as "master" instead of "adhoc". Only the "activated" state is shown correctly.

The settings in /etc/config/wireless are OK (unless you press "submit" on a page with wrongly displayed values).

SvenRoederer, 2017-12-28:
can you check with vanilla LEDE 17.01.4  when running the same wirless config there (manually configured)?

torte71, 2017-12-29:
OK, it is in vanilla LEDE-17.01.4 as well - and seems known: https://github.com/openwrt/luci/issues/1179

torte71, 2018-01-13:
It seems halfway fixed in openwrt/luci: The name is shown correctly, but the mode is still wrong.
They rearranged their code so it is not easy to include their changes 1:1, because it leads to circular "require"s.

So we might
- leave it as is and get the new code, when we update to a newer openwrt revision
  (though it means wild guessing, if you have several disabled APs, cause they all look the same)
- include a more recent luci feed
  (but that could be a little risky, if we want to release 1.0.0 soon)
- wait, until someone has a better idea
- use a quick hack like this to display the correct SSID and mode:

```
diff --git a/modules/luci-base/luasrc/model/network.lua b/modules/luci-base/luasrc/model/network.lua
index 49d91b8..938ba74 100644
--- a/modules/luci-base/luasrc/model/network.lua
+++ b/modules/luci-base/luasrc/model/network.lua
@@ -1550,6 +1550,7 @@ end

 function wifinet.active_mode(self)
 	local m = self.iwinfo.mode or self:ubus("net", "config", "mode") or self:get("mode") or "ap"
+	if not self:is_up() then m = self:get("mode") or "ap" end

	if     m == "ap"      then m = "Master"
	elseif m == "sta"     then m = "Client"
@@ -1566,6 +1567,7 @@ function wifinet.active_mode_i18n(self)
 end

 function wifinet.active_ssid(self)
+	if not self:is_up() then return self:get("ssid") end
 	return self.iwinfo.ssid or self:ubus("net", "config", "ssid") or self:get("ssid")
 end
 
```

bobster-galore, 2018-01-13:
I can confirm this issue for cpe210 @ hedy 1.0.0 rc.
Fortunately I disabled only one adhoc-network, so I can easily distinguish. Since we want to push Mesh (11s) may be an intermediate solution like @torte71 sketched would be helpful for others who diasble their adhoc-networks.

SvenRoederer, 2018-01-13:
@torte71 have you tried to cherry-pick this into luci 17.04?

torte71, 2018-01-13:
@SvenRoederer : Not yet, the current code there is quite different (and well, I need to learn more about git, e.g. "cherry pick" first).

SvenRoederer, 2018-01-14:
[2] have you tried luic-head on a LEDE-17.01-bsed system? As Luci-head is matching to LEDE-snapshot, where even the Luci-core / libs, iwinfo and others (remember the problems with owm on SAm0815-experimental builds) have changed, that might the reason the Patch is only working on Luci-head.

[3] have in mind, that the Luci used in 17.04 is commit https://github.com/openwrt/luci/commit/d3f0685d63c1291359dc5dd089c82fa1e150e0c6. there are some more changes up to the commit used in Hedy-1.0.0-rc1 (https://github.com/openwrt/luci/commit/d2dc32af312097fd9588b8d193308c7d7459a9ca)
```
sven@build:/usr/src/openwrt/packages_sven/openwrt-luci$ git log --oneline d3f0685d63c1291359dc5dd089c82fa1e150e0c6..d2dc32af312097fd9588b8d193308c7d7459a9ca|less
d2dc32af3 luci-app-statistics: set logfile as first item in collectd.conf
a9bddc408 timezone data: update to 2017c
07e7ff07a Fixes #1413.
b74ef26be Merge pull request #1411 from Cye3s/lede-17.01
4a0877b4e luci_base: Update zh_TW base.po, translate fix
1aae4e9c8 luci_base: Update zh_CN base.po, translate fix
0ed808690 Merge pull request #1403 from StevenHessing/lede-17.01-noddos
d685c2392 luci-app-noddos: backport from master
7fc88b4e6 luci-mod-admin-full: move KRACK countermeasures out of IEEE802.11w condition
```
But even this seems not related. 

[4] Probably something was changed on LEDE itself since "17.01-4". see [3]

torte71, 2018-01-14:
[2] I just installed the LEDE-snapshot from today: https://downloads.lede-project.org/snapshots/targets/ar71xx/generic/ and the required luci,lua,... packages:
In this build, even the "mode" is displayed correctly.
- After reverting the changes from [2] "luci-base: fix luci.sys.wifi.getiwinfo() on radio names", still everything is displayed correctly (and yes, I cleared /tmp/luci-modulecache/*). No idea, what special case that change should handle.
- Code in network.lua (from [4]) looks unchanged.
- iwinfo.mode is now nil, too (leading to a correct display)
```
LEDE-snapshot
  wifinet.active_mode()
    ssid:"AP2adhoc"
    iwinfo.mode:nil
    ubus:mode:nil
    self.mode:"adhoc"

  wifinet.active_ssid()
    ssid:"AP2adhoc"
    iwinfo.ssid:nil
    ubus:ssid:nil
    self.ssid:"AP2adhoc"
```

==>I think, the different behaviour is not coming from the .lua files, but rather from the underlying system libraries.

To check, that this is not hardware dependent, I repeated the Lede-17.01.4 test on exactly the same device, that I've used for the Lede-Snapshot test: The results for 17.01.4 are identical on different devices (ssid ok, mode wrong), whereas the results from different Lede-Versions on the same device show different results (17.01.4: ssid ok, mode wrong <-> Snapshot: ssid and mode ok).

Devices used for tests:
- Hedy-1.0.0-rc1: gl-ar150
- Lede-17.01.4: wr841v9
- Lede-Snapshot: wr842v3
- Lede-17.01.4 (second attempt): wr842v3

SvenRoederer, 2018-01-15:
So it'S probably the best to use the quick-fix you outlined in https://github.com/freifunk-berlin/firmware/issues/501#issuecomment-357460747 for the upcoming release.

torte71, 2018-01-15:
Btw.: These are (some of) the changes to iwinfo between Hedy-1.0.0-rc1 (2016-09-21; fd9e17be0c43bd6b8df5371f0b353747bc563874) and OpenWrt/Head (2017-08-23; c1a03e8231a5d8b348b70a182d256725c98a3b0b):
https://git.openwrt.org/?p=project/iwinfo.git;a=commitdiff;h=ab089dda34a44bd118180a9403abb942b255ad2e;hp=335967c6c210c34edc07a54849230314fc049143

See esp. iwinfo_nl80211.c: nl80211_get_mode() and nl80211_get_ssid():
Taking care of return values might turn out as a good habit :)

SvenRoederer, 2018-01-16:
I'd like to ask, if it's so important to have this display fix (which was like it is since many years) in this release? 
If it's an easy fix it might be okay  to correct, but if if becomes more messy at every step - it it worth to fix something which is fixed upstream when switching to the next release?

Probably it's more effective to spend this energy in the "experimental"-branch to go froward then backporting to lede-17.04 (Hedy-1.0.0)?

torte71, 2018-01-17:
That might indeed be better.

I've build a 1.0.0-rc1 version with the git-HEAD version of iwinfo, it compiles without a hazzle, but the displayed values are still wrong.
Btw., the 0.3.0 behaves correctly.

As it can be worked-around by editing network.lua on the router and probably not many people will have lots of disabled APs, those few that are annoyed by this behaviour can apply an ugly hack, if they like.

jow-, 2018-07-22:
https://github.com/openwrt/luci/commit/f9b8d7ff7bbd3c82f9889b65af07a28d94d9fed7 is responsible for fixing the leak of wireless state info across vifs.

SvenRoederer, 2018-07-23:
@jow- correct in a recent build based in LuCI-master the wireless-signal strength seems fine, even when no client is connected.

[image]

pmelange, 2018-11-04:
I have just tested this with the master branch.  Everything seems to be displayed fine.

Has this been fixed upstream or does it work only because of the patch?

SvenRoederer, 2019-12-18:
@torte71 last time I was seeing some deactivated APs on the wireless-page , their symbols looked fine.
Do you see any issues?

torte71, 2019-12-18:
I just checked these cpe220-v1 default images from buildbot:

69 "master", revision 1b49195: ok
70 "Hedy-1.0.x", revision 6649c99: not fixed
74 "Hedy-1.0.x_kernel4.4.205", revision 87dc6b3: untestable: permanent reboots
76 "upstream-1701", revision 2e2d829: not fixed; reboots sometimes
78 "upstream-1907", revision f74ad18: ok

("not fixed" means: Wrong settings, like a wrong SSID from a different virtual AP, are displayed)

SvenRoederer, 2019-12-18:
So fixed for master and all OpenWrt-19.07 and newer branches. For Hedy-1.0.x (Lede-17.01) the 
commit mentioned by @jow- applies cleanly and I will start a  build for this (8057733a60).

@torte71  regarding the permanent reboots, please check https://github.com/freifunk-berlin/firmware/commit/87dc6b324cc6dacfea3ebaed8052e6cd0072b418 or open an issue

SvenRoederer, 2019-12-19:
just started a new build, with default-kernel, but luci-wifi-patch (7e96a9706e)

SvenRoederer, 2019-12-20:
with this patch applied the wifi-setup page is broken here. It stucks with "collecting data" and gives some lua-errors:
> Fri Dec 20 23:25:28 2019 daemon.err uhttpd[2303]: /usr/lib/lua/luci/dispatcher.lua:380: Failed to execute call dispatcher target for entry '/admin/network/wireless_status/radio0.network1,radio0.network2,radio0.network3'.
Fri Dec 20 23:25:28 2019 daemon.err uhttpd[2303]: The called action terminated with an exception:
Fri Dec 20 23:25:28 2019 daemon.err uhttpd[2303]: /usr/lib/lua/luci/model/network.lua:187: attempt to call global '_wifi_state_by_ifname' (a nil value)
Fri Dec 20 23:25:28 2019 daemon.err uhttpd[2303]: stack traceback:
Fri Dec 20 23:25:28 2019 daemon.err uhttpd[2303]:       [C]: in function 'assert'
Fri Dec 20 23:25:28 2019 daemon.err uhttpd[2303]:       /usr/lib/lua/luci/dispatcher.lua:380: in function 'dispatch'
Fri Dec 20 23:25:28 2019 daemon.err uhttpd[2303]:       /usr/lib/lua/luci/dispatcher.lua:109: in function </usr/lib/lua/luci/dispatcher.lua:108>

I've no real interest in investing time for fixing this for v1.0.x releases (Lede-17.01 based). It's fixed upstream in the more recent branches (>= OpenWrt 18.06), so I'm fine with closing unfixed and moving on.
@torte71 if you like to see this fixed, feel free to send a PR.

SvenRoederer, 2021-06-13:
I'll close, as firmware is based on OpenWrt-21.02 now where the issue is fixed (https://github.com/freifunk-berlin/firmware/issues/501#issuecomment-567246904) and the problematic Lede-17.01 based Hedy is EOL.

closed 2021-06-13 by SvenRoederer