freifunk-berlin/firmware #502: sam0815-43478e9-ubnt-nano-m: no internet

lightcatcher, 2017-12-27:
betrifft: 572-freifunk-berlin-1.0.0-alpha-sam0815-43478e9-ubnt-nano-m
Nanostation M2, frisch eingerichtet, mehrfach versucht.

Kein internet via lan+wlan, mesh partner sehen sich.
ping vom gerät zu lede-project.org gibts "bad adress"
log nach einrichtung +reboot im anhang.

[no internet-nano-m2.txt](https://github.com/freifunk-berlin/firmware/files/1589058/no.internet-nano-m2.txt)

SvenRoederer, 2017-12-31:
* Is the NSM seeing the OLSR-neighbours?
* what's the output of ```ip route get 8.8.8.8```?

lightcatcher, 2017-12-31:
NSM sees the wdr3600 in olsr neighbours

root@berlinerstr-tegel-m2-2:~# ip route get 8.8.8.8
8.8.8.8 dev tnl_0a1f0057  src 10.31.15.28 
    cache 
-----

lightcatcher, 2018-01-01:
now working with Hedy-1.0.0-rc1.

closed 2018-01-01 by lightcatcher