freifunk-berlin/firmware #504: LEDE build: Several OLSR pages in admin mode broken

torte71, 2017-12-31:
https://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/578/VERSION.txt
Firmware: git branch "SAm0815_experimental", revision f6104a8
Device independent (also on rpi3 and gl-ar150).

When logged in as admin, the pages under "status"/"olsr" show missing values or throw out errors:
- OLSR/Overview: http://frei.funk/cgi-bin/luci/admin/status/olsr
No values displayed (only "-"), though there are connected mesh partners

- OLSR/Interfaces: http://frei.funk/cgi-bin/luci/admin/status/olsr/interfaces
Interfaces are listed, but displayed as "not started" (smells like being related to missing values in the overview page)

- OLSR/MID: http://frei.funk/cgi-bin/luci/freifunk/olsr/mid
Displays an error message:
`/usr/lib/lua/luci/dispatcher.lua:390: Failed to execute call dispatcher target for entry '/freifunk/olsr/mid'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/olsr.lua:325: attempt to compare two nil values
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:390: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>`

- OLSR/Routes: http://frei.funk/cgi-bin/luci/admin/status/olsr/routes
Displays a different error message:
`/usr/lib/lua/luci/dispatcher.lua:390: Failed to execute call dispatcher target for entry '/admin/status/olsr/routes'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/olsr.lua:398: attempt to index local 'jsondata4' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:390: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>`

I've already corrected the plugins version numbers in /etc/config/olsrd[6] to match the files in /usr/lib/, so this is not the cause.

torte71, 2018-01-10:
Some progress:

The behaviour of the olsr-jsoninfo plugin has changed between olsrd-0.9.0.3 and 0.9.6.2: The "status" keyword does no longer work ("echo /status | nc 127.0.0.1 9090").

With 0.9.6.2, it outputs {"pid": 2573,"systemTime": 1515602775,"timeSinceStartup": 1972320,"configurationChecksum": "d9069f28","error": "404 Not Found"}

With 0.9.0.3, all of "/status", "/all" and "/blablabla"(sic!) give the same output (differing in timestamps only).
 
You won't find any reference to "/status" in the plugin source: https://github.com/OLSR/olsrd/blob/master/lib/jsoninfo/src/olsrd_jsoninfo.c#L127
This is the same for the 0.9.0.3 plugin. No "/status" is mentioned. It could only have worked through some kind of fall-through/catch-all logic (intended or not).
Only reference is here :) : http://www.olsr.org/?q=jsoninfo_plugin

I changed /usr/lib/lua/luci/controller/olsr.lua
```
--      jsonreq4 = utl.exec("(echo /status | nc 127.0.0.1 " .. v4_port .. " | sed -n '/^[}{ ]/p') 2>/dev/null" )
        jsonreq4 = utl.exec("(echo /all | nc 127.0.0.1 " .. v4_port .. " | sed -n '/^[}{ ]/p') 2>/dev/null" )
--      jsonreq6 = utl.exec("(echo /status | nc ::1 " .. v6_port .. " | sed -n '/^[}{ ]/p') 2>/dev/null")
        jsonreq6 = utl.exec("(echo /all | nc ::1 " .. v6_port .. " | sed -n '/^[}{ ]/p') 2>/dev/null")
```
And now I get numbers on the OLSR status page.

SvenRoederer, 2018-01-10:
think this an an dup of https://github.com/freifunk-berlin/firmware/issues/444

SvenRoederer, 2018-01-10:
As olsrd 0.9.6+ seems to cause problemms in the net (at least with BBB-VPN - olsrd pre-0.6.) we decided to stay with the 0.9.0.3 version for now.

closed 2018-05-01 by SvenRoederer