freifunk-berlin/firmware #505: LEDE build: "diff" missing

torte71, 2017-12-31:
https://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/576/VERSION.txt
Firmware: git branch "SAm0815_experimental", revision 1ddb59a

I want my "diff" back :)

`diff network.bak network`
`-ash: diff: not found`

`opkg update && opkg search "*diff*"`
`[ no results ]`

Would be nice to have at least an opkg-installable package.

SvenRoederer, 2017-12-31:
it has never been here, but you can find the ```diffutils``` in the packages feed.
so you can install from the original LEDE-repo or send PR to enable in our builds

SvenRoederer, 2018-01-29:
will you do a PR for this?

torte71, 2018-01-31:
OK. I'll add diffutils and patch as optional packages (i.e. not included in the images).
Will a single PR for "head" suffice? It would apply (with fuzz) and work with rc1 and experimental as well.

closed 2018-02-03 by torte71