freifunk-berlin/firmware #506: LEDE build: ssh-login banner lacks "nice" build name

torte71, 2017-12-31:
https://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/578/VERSION.txt
Firmware: git branch "SAm0815_experimental", revision f6104a8

OK, this is not really a showstopper, but anyway:
When logging in with ssh, the banner displays
`Firmware Berlin (%N 1.1.0-alpha-SAm0815 rev 1ddb59a)`

Shouldn't that mention hedy?
`Firmware Berlin (hedy 1.1.0-alpha-SAm0815 rev 1ddb59a)`

SvenRoederer, 2017-12-31:
28bd2179992a73f94f7631d7c7ecd88bfe87037b probably causing this.
The upstream banner file is using ```%D %V, %C``` so %D seems better than %N. But the meaning of the placeholders can be checked somewhere in "make menuconfig"

SvenRoederer, 2018-01-22:
this is caused by the upstream change described in issue #515

SvenRoederer, 2018-01-29:
@torte71 do you agree that this issue seems to be cause by the problem noted on issue #515? 
Then please close this issue an we only take care of it in the other one.

torte71, 2018-01-30:
OK

closed 2018-01-30 by torte71