freifunk-berlin/firmware #507: LEDE build: Devices not listed on openwifimap

torte71, 2017-12-31:
https://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/578/VERSION.txt
Firmware: git branch "SAm0815_experimental", revision f6104a8

Even though location settings are correctly entered in the wizard page (and made sure, the lat/lon values are not mixed up to meet some berlin friends in the arabian sea), the device doesn't get listed on https://openwifimap.net/
The device is running for over a day, so I think that should be enough time for an update?

SvenRoederer, 2017-12-31:
What is the output when you run "owm.lua" from the command-line?

torte71, 2017-12-31:
On Sun, 31 Dec 2017 04:47:03 -0800
Sven Roederer <notifications@github.com> wrote:

> What is the output when you run "owm.lua" from the command-line?

(run on a gl-ar150 - if that matters)

/usr/bin/lua: ?:0: attempt to call field 'arptable' (a nil value)
stack traceback:
        ?: in function 'get'
        /usr/sbin/owm.lua:74: in main chunk
        [C]: ?

SvenRoederer, 2017-12-31:
seems like the LUCI-Api has changed again (see https://github.com/freifunk-berlin/firmware-packages/pull/109 for something similar).
The current error causes the owm-script to fail to run and so there will be no upload to the server.

error comes from https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-owm/files/owm.lua#L74, which is very likely causes by 
https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-owm/luasrc/owm.lua#L447 and following

SvenRoederer, 2018-03-06:
sys.net.arptable() call in https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-owm/luasrc/owm.lua#L447  has changed.
Changing the line to 
> local arptable = {} 
makes the code run, but without the values.

SvenRoederer, 2018-06-03:
@ikreb7 the file is located in /usr/lib/lua/luci as per https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-owm/Makefile#L98

SvenRoederer, 2018-08-20:
Btw. do we really need the anonymized MAC-adresses in the public OWM-data?
What's their purpose?

booo, 2018-08-21:
Some functions where dropped in the luci code base with this commit this https://github.com/openwrt/luci/commit/366707a681459a4d520dc97024ea0a4b3c24a326. 

> luci.sys.net.arptable()   - replaced by luci.ip.neighbors()

Maybe it's enough to replace the function call?

booo, 2018-08-23:
I merged a commit that should fix the problem.

We still have to update the feed for the berlin packages in the firmware.

SvenRoederer, 2018-08-23:
still fails for me with
> /usr/bin/lua: /usr/lib/lua/luci/owm.lua:447: attempt to index global 'luci' (a nil value)
stack traceback:
        /usr/lib/lua/luci/owm.lua:447: in function 'get'
        /usr/sbin/owm.lua:74: in main chunk
        [C]: ?

SvenRoederer, 2018-08-23:
> /usr/bin/lua: /usr/lib/lua/luci/owm.lua:447: attempt to index global 'luci' (a nil value)
is fixed by https://github.com/freifunk-berlin/firmware-packages/commit/c883485708a795be3fd34f220b373faee42e4f06.

the code now fails with 
> /usr/bin/lua: /usr/lib/lua/luci/owm.lua:452: attempt to index field 'HW address' (a nil value)
stack traceback:
        /usr/lib/lua/luci/owm.lua:452: in function 'get'
        /usr/sbin/owm.lua:74: in main chunk
        [C]: ?

booo, 2018-08-24:
Sorry for the inconvenience. Seems like my testing procedure was flawed.

I will prepare a patch for the outstanding problems. We have to change the field names and convert them from userdata to string.

booo, 2018-09-15:
As far as I can see we fixed this.

closed 2018-09-15 by booo