freifunk-berlin/firmware #508: rpi-3: Allow only channels 1 to 11

kehugter, 2018-01-02:
Vanilla LuCi in LEDE 17.01.4 allows only channels 1 to 11. Tested with `lede-17.01.4-brcm2708-bcm2710-rpi-3-ext4-sdcard.img`.

People on Raspbian use the `next` branch from https://github.com/Hexxeh/rpi-firmware, which runs a 4.14 kernel, to get channels 12 and 13 working. In the future we might have them too.

SvenRoederer, 2018-01-03:
when this can also be seen upstream, it's the best to report there and also fix there

kehugter, 2018-01-03:
It is fixed there, the problem is here: default installation channel is 13, channel selector in LuCi allows up to 13.

SvenRoederer, 2018-01-03:
didn't you mention vanilla LEDE is also affected?

kehugter, 2018-01-03:
Maybe I didn't phrase it right. Channels 12 and 13 don't work yet for the `rpi-3`. Vanilla LuCI allows selecting channels up to 11. Right now, the berlin firmware defaults to channel 13 and LuCI has a channel selector going up to 13.

bobster-galore, 2018-01-03:
So there should be an rpi-3 freifunk flavour defaulting to channel 1-11?
At least until channel 12-13 are available for rpi-3?

SvenRoederer, 2018-01-03:
This will make meshing with "regular" nodes impossible, so no option to go for freifunk.

@torte71 @kehugter btw. what is the WiFi-chip used on the board?

kehugter, 2018-01-03:
@SvenRoederer Broadcom BCM43438

https://wikidevi.com/wiki/RPF_Raspberry_Pi_3_Model_B

Seems to be supported by brcmfmac (https://wireless.wiki.kernel.org/en/users/Drivers/brcm80211). IIRC, the non-free drivers are present in LEDE.

torte71, 2018-01-03:
This is, what the debian guys write about the closed/open source drivers on rpi3: https://wiki.debian.org/RaspberryPi3

SvenRoederer, 2018-01-03:
I remember Broadcom-drivers have never been fun.

but when it fails under LEDE-17.04 did you also test with LEDE-Snapshot?

torte71, 2018-01-03:
Yesterday I checked the current snapshot from the lede buildbot. Only single AP, channel switching only after reboot, no channel 12 or 13.
Anyway, it now runs as a ff/tunnelberlin-router at channel 6 at work everyday to get some long term results. Of course without mesh.

kehugter, 2018-01-03:
> Anyway, it now runs as a ff/tunnelberlin-router at channel 6 at work everyday to get some long term results. Of course without mesh.

This is also good enough for me. There aren't any nodes in my vicinity, so I'd be just sharing internet.

> IIRC, the non-free drivers are present in LEDE

Already removed in LEDE 17
https://github.com/lede-project/source/commit/c3d3111831d49d1ab37d90bbe6a61f1b2263721d

> At least until channel 12-13 are available for rpi-3?

LEDE master branch is using a 4.14 kernel, the issue should be solved by the next major release.

kehugter, 2018-01-03:
I could use channel 11 but not 13 in https://github.com/lede-project/source/commit/c5ca1c9ab65bfe1e6fc74230f8c0121230562b1c, I'll see if there's an issue open to fix it.