freifunk-berlin/firmware #509: SAm0815_Hedy-alpha: No Wifi LEDs

carsifreifunk, 2018-01-07:
Wifi LEDs bleiben aus, bei aktivierten WLAN.

bobster-galore, 2018-01-07:
Weird I've got Hedy 1.0.0-rc1 7f42122 default und w/o vpn running on two CPR210 V1.1 with fully functional leds.

bobster-galore, 2018-01-07:
Could u pls check /etc/config/system.
My config is like that:
```
config led 'led_lan0'
	option name 'LAN0'
	option sysfs 'tp-link:green:lan0'
	option trigger 'switch0'
	option port_mask '0x20'

config led 'led_lan1'
	option name 'LAN1'
	option sysfs 'tp-link:green:lan1'
	option trigger 'switch0'
	option port_mask '0x10'

config rssid 'rssid_wlan0'
	option dev 'wlan0'
	option refresh '40000'
	option threshold '1'

config led 'led_rssilow'
	option name 'RSSILOW'
	option sysfs 'tp-link:green:link1'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '1'
	option maxq '100'
	option offset '0'
	option factor '13'

config led 'led_rssimediumlow'
	option name 'RSSIMEDIUMLOW'
	option sysfs 'tp-link:green:link2'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '26'
	option maxq '100'
	option offset '-25'
	option factor '13'

config led 'led_rssimediumhigh'
	option name 'RSSIMEDIUMHIGH'
	option sysfs 'tp-link:green:link3'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '51'
	option maxq '100'
	option offset '-50'
	option factor '13'

config led 'led_rssihigh'
	option name 'RSSIHIGH'
	option sysfs 'tp-link:green:link4'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '76'
	option maxq '100'
	option offset '-75'
	option factor '13'
```

SvenRoederer, 2018-01-23:
Just reflashed my CPe210 with revision ab3754cb311c84fb99adc61ab583ec0198510ecc and also have no LED-activity.
'/etc/config/system' looks similar
```
config rssid 'rssid_wlan0'
        option dev 'wlan0'
        option refresh '200000'
        option threshold '1'

config gpio_switch 'poe_passthrough'
        option name 'PoE Passthrough'
        option gpio_pin '20'
        option value '0'

config led 'led_lan0'
        option name 'LAN0'
        option sysfs 'tp-link:green:lan0'
        option trigger 'switch0'
        option port_mask '0x20'

config led 'led_lan1'
        option name 'LAN1'
        option sysfs 'tp-link:green:lan1'
        option trigger 'switch0'
        option port_mask '0x10'

config led 'led_rssilow'
        option name 'RSSILOW'
        option sysfs 'tp-link:green:link1'
        option trigger 'rssi'
        option iface 'rssid_wlan0'
        option minq '1'
        option maxq '100'
        option offset '0'
        option factor '13'

config led 'led_rssimediumlow'
        option name 'RSSIMEDIUMLOW'
        option sysfs 'tp-link:green:link2'
        option trigger 'rssi'
        option iface 'rssid_wlan0'
        option minq '26'
        option maxq '100'
        option offset '-25'
        option factor '13'

config led 'led_rssimediumhigh'
        option name 'RSSIMEDIUMHIGH'
        option sysfs 'tp-link:green:link3'
        option trigger 'rssi'
        option iface 'rssid_wlan0'
        option minq '51'
        option maxq '100'
        option offset '-50'
        option factor '13'

config led 'led_rssihigh'
        option name 'RSSIHIGH'
        option sysfs 'tp-link:green:link4'
        option trigger 'rssi'
        option iface 'rssid_wlan0'
        option minq '76'
        option maxq '100'
        option offset '-75'
        option factor '13'
```

But 'ps w' showed the rssileds process and parameters with 
> /usr/sbin/rssileds wlan0 200000 1 tp-link:green:link1 1 100 0 13 ...
and I know per default there is no wlan0 interface in the firmware. I manually started the precess with
> /usr/sbin/rssileds wlan0-adhoc-2 200000 1 tp-link:green:link1 1 100 0 13 ....
and 3 of 4 LEDs came on.

I was not able to test if they change by signal-strength ...

carsifreifunk, 2018-01-24:
Ich habe noch mal ein Update auf ein Hedy von heute gemacht. Keine WIfi LED an. 

Auszug aus /etc/config/system
config system
	option ttylogin '0'
	option log_size '64'
	option urandom_seed '0'
	option cronloglevel '10'
	option zonename 'Europe/Berlin'
	option timezone 'CET-1CEST,M3.5.0,M10.5.0/3'
	option hostname '254-206-hedytest'

config timeserver 'ntp'
	option enabled '1'
	option enable_server '0'
	list server '0.lede.pool.ntp.org'
	list server '1.lede.pool.ntp.org'
	list server '2.lede.pool.ntp.org'
	list server '3.lede.pool.ntp.org'

config rssid 'rssid_wlan0'
	option dev 'wlan0'
	option refresh '200000'
	option threshold '1'

config gpio_switch 'poe_passthrough'
	option name 'PoE Passthrough'
	option gpio_pin '20'
	option value '0'

config led 'led_lan0'
	option name 'LAN0'
	option sysfs 'tp-link:green:lan0'
	option trigger 'switch0'
	option port_mask '0x20'

config led 'led_lan1'
	option name 'LAN1'
	option sysfs 'tp-link:green:lan1'
	option trigger 'switch0'
	option port_mask '0x10'

config led 'led_rssilow'
	option name 'RSSILOW'
	option sysfs 'tp-link:green:link1'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '1'
	option maxq '100'
	option offset '0'
	option factor '13'

config led 'led_rssimediumlow'
	option name 'RSSIMEDIUMLOW'
	option sysfs 'tp-link:green:link2'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '26'
	option maxq '100'
	option offset '-25'
	option factor '13'

config led 'led_rssimediumhigh'
	option name 'RSSIMEDIUMHIGH'
	option sysfs 'tp-link:green:link3'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '51'
	option maxq '100'
	option offset '-50'
	option factor '13'

config led 'led_rssihigh'
	option name 'RSSIHIGH'
	option sysfs 'tp-link:green:link4'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '76'
	option maxq '100'
	option offset '-75'
	option factor '13'

SvenRoederer, 2018-01-25:
@bobster-galore can you check :
* the parameters given to "rssileds"?
* if you have a "wlan0" interface defined ("iwinfo")

bobster-galore, 2018-01-25:
```
config system
	option ttylogin '0'
	option log_size '64'
	option urandom_seed '0'
	option version '1.0.0'
	option cronloglevel '10'
	option zonename 'Europe/Berlin'
	option timezone 'CET-1CEST,M3.5.0,M10.5.0/3'
	option hostname 'cpe-spd02'
	option longitude '13.215000000000000'
	option location 'Spandau'
	option latitude '52.548478'

config timeserver 'ntp'
	option enabled '1'
	option enable_server '0'
	list server '0.lede.pool.ntp.org'
	list server '1.lede.pool.ntp.org'
	list server '2.lede.pool.ntp.org'
	list server '3.lede.pool.ntp.org'

config rssid 'rssid_wlan0'
	option dev 'wlan0'
	option refresh '200000'
	option threshold '1'

config gpio_switch 'poe_passthrough'
	option name 'PoE Passthrough'
	option gpio_pin '20'
	option value '0'

config led 'led_lan0'
	option name 'LAN0'
	option sysfs 'tp-link:green:lan0'
	option trigger 'switch0'
	option port_mask '0x20'

config led 'led_lan1'
	option name 'LAN1'
	option sysfs 'tp-link:green:lan1'
	option trigger 'switch0'
	option port_mask '0x10'

config led 'led_rssilow'
	option name 'RSSILOW'
	option sysfs 'tp-link:green:link1'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '1'
	option maxq '100'
	option offset '0'
	option factor '13'

config led 'led_rssimediumlow'
	option name 'RSSIMEDIUMLOW'
	option sysfs 'tp-link:green:link2'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '26'
	option maxq '100'
	option offset '-25'
	option factor '13'

config led 'led_rssimediumhigh'
	option name 'RSSIMEDIUMHIGH'
	option sysfs 'tp-link:green:link3'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '51'
	option maxq '100'
	option offset '-50'
	option factor '13'

config led 'led_rssihigh'
	option name 'RSSIHIGH'
	option sysfs 'tp-link:green:link4'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '1'
	option maxq '100'
	option offset '0'
	option factor '13'

config led 'led_rssimediumlow'
	option name 'RSSIMEDIUMLOW'
	option sysfs 'tp-link:green:link2'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '26'
	option maxq '100'
	option offset '-25'
	option factor '13'

config led 'led_rssimediumhigh'
	option name 'RSSIMEDIUMHIGH'
	option sysfs 'tp-link:green:link3'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '51'
	option maxq '100'
	option offset '-50'
	option factor '13'

config led 'led_rssihigh'
	option name 'RSSIHIGH'
	option sysfs 'tp-link:green:link4'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '76'
	option maxq '100'
	option offset '-75'
	option factor '13'
```
"wlan0" interface is defined
```
iwinfo
wlan0     ESSID: unknown
          Access Point: 00:00:00:00:00:00
          Mode: Mesh Point  Channel: 13 (2.472 GHz)
          Tx-Power: 20 dBm  Link Quality: 70/70
          Signal: -40 dBm  Noise: -83 dBm
          Bit Rate: 78.3 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan0-dhcp-2 ESSID: "berlin02.freifunk.net"
          Access Point: 62:E3:27:CD:63:D8
          Mode: Master  Channel: 13 (2.472 GHz)
          Tx-Power: 20 dBm  Link Quality: 44/70
          Signal: -66 dBm  Noise: -83 dBm
          Bit Rate: 9.4 MBit/s
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

SvenRoederer, 2018-01-25:
that fits into the picture drawn by me.
@bobster-galore  if you rename the 802.11s-interface to something like "wlan0-mesh-2" I assume that you will also have no LEDs

It this is true, we need to check for the syntax of system.led_rssi[*].ifname, to act on our "wlan0-adhoc-2"

bobster-galore, 2018-01-25:
The name for the mesh-interface was set by luci when I set the mesh up, how can I change that within luci?

SvenRoederer, 2018-01-25:
@bobster-galore at least for AP-Mode I've seen this in the advanced options. Or check for something like ifname in uci->wireless

bobster-galore, 2018-01-25:
Ok, overwrote in luci/wireless/advanced settings/interfacename with wlan0-mesh-2 -> lights went off except power and lan0 signalling traffic. 
no sign of renamed interface in '/etc/config/system'.
Just checked kathleen 0.2.0
```
config rssid 'rssid_wlan0'
	option dev 'wlan0'
	option refresh '40000'
	option threshold '1'
```
urban legend was that these leds wont be on because as an ap it can't show signal strength. But as we see above there has never been an interface called wlan0, so no reason to shine.

After overwriting if-name again with wlan0 leds are on again. 
But what information do we get from 4 green leds?

SvenRoederer, 2018-01-25:
The point in monitoring the rssi is, do the current LED-pattern reflect the received signal-level?
Or other way around: is a low signal causing the LED-pattern to change?

bobster-galore, 2018-01-25:
I switched off all other routers, i.e. no mesh partner. 4 bright leds - not such much change.

SvenRoederer, 2018-01-26:
probably changing the values of 
> option minq '76'
option maxq '100'
option offset '-75'
option factor '13'

will help?

On my node I changed a bit the "minq" values, let's wait till it's dark outside ...

bobster-galore, 2018-01-27:
Seems to work at gluon https://github.com/freifunk-gluon/gluon/issues/410 .

SvenRoederer, 2018-01-27:
I was able to switch between all LEDs on and all LEDS of.
Probably a bit of verbose printing what "rssiled" daemon is doing internally will help to fine tune the values.

The commit references in Gluon https://github.com/freifunk-gluon/gluon/commit/f5764a4a7b68f2eb56c9c5f2995a6277bf1b4aaa is just changing the device from "wlan0" to "mesh", which is the same I did ("wlan0-mesh-2" or "wlan0-adhoc-2")

SvenRoederer, 2018-01-29:
We can change the rssi-device to "wlan0-adhoc-2" for Hedy-1.0.0 but as reported this will only cause the LEDs to stay on all the time. This seems the opposite option of the reported issue.
And both states (all-on and all-off) present the same non-information. So I'll keep this open and postpone to a later release.
This also gives that chance to combine a fix of this with the switch-over to 802.11s (and updated ifnames)

bobster-galore, 2018-09-21:
To me it seems to be not mission critical, may we postpone it to a later milestone until some time is found to fix it?