freifunk-berlin/firmware #511: SAm0815_experimental: Raspberry PI3 kernel build failure

torte71, 2018-01-18:
When building rpi-3 (i.e. brcm2708_bcm2710) images with kernels newer than 4.4.x (hedy-1.0.0-rc1+openwrt-17.01 use  4.4.x; hedy-experimental+openwrt-head use 4.9.x), our default setting
``CONFIG_STRIP_KERNEL_EXPORTS=y``
breaks the build process. It leads to a (kernel-)dependency of "symtab.h", which does not get generated in the arm64 builds.

To get around that, the file "configs/brcm2708-bcm2710.config" needs to have following line appended:
`` CONFIG_STRIP_KERNEL_EXPORTS=n``

The 32bit brcm2708_2708 image behaves differently: It creates the "symtab.h", so there is no need modifying its .config file. But I don't have a clue yet, why the "symtab.h" is generated on arm, but not on arm64. Still investigating...

The following output is from a former build (based on lede), but the same happens on actual versions (based on openwrt^56767dfa4232d3045136cf6c02956bfd4451af5e):
```
make -C /home/torte/openwrt/hedy/SAm0815_experimental-brcm2708-bcm2710/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/linux-4.9.72 HOSTCFLAGS="-O2 -I/home/torte/openwrt/hedy/SAm0815_experimental-brcm2708-bcm2710/lede/staging_dir/host/include -I/home/torte/openwrt/hedy/SAm0815_experimental-brcm2708-bcm2710/lede/staging_dir/host/usr/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Dec 30 13:17:34 2017" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/torte/openwrt/hedy/SAm0815_experimental-brcm2708-bcm2710/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.9.72 EXTRA_LDSFLAGS="-I/home/torte/openwrt/hedy/SAm0815_experimental-brcm2708-bcm2710/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710 -include symtab.h" CC="aarch64-openwrt-linux-musl-gcc" modules
make[6]: Entering directory '/home/torte/openwrt/hedy/SAm0815_experimental-brcm2708-bcm2710/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/linux-4.9.72'
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  LDS     arch/arm64/kernel/vdso/vdso.lds
cc1: fatal error: symtab.h: No such file or directory
compilation terminated.
```

torte71, 2018-01-18:
This is an upstream issue: https://bugs.lede-project.org/index.php?do=details&task_id=1282

As memory usage is not exactly a problem on raspberries, we may consider adding the above option to brcm2708-bcm2710.config until it gets fixed upstream.

SvenRoederer, 2018-01-18:
As 'CONFIG_STRIP_KERNEL_EXPORTS' I agree that it seems to be upstream, as it don't cause a build error on other platforms.

But sure, we can add this option temporarly fix for brcm2708-bcm2710 on the develop/master branch.

SvenRoederer, 2018-02-24:
this seems to be fixed by https://github.com/openwrt/openwrt/pull/696

torte71, 2018-02-24:
Not really... https://github.com/openwrt/openwrt/pull/696#issuecomment-368224478

pmelange, 2018-11-04:
Is this still an issue?

torte71, 2018-12-21:
No changes yet (tested master 32ccf34):
```
  LDS     arch/arm64/kernel/vdso/vdso.lds
cc1: fatal error: symtab.h: No such file or directory
compilation terminated.
```
The option "CONFIG_STRIP_KERNEL_EXPORTS=n" is still required.

pmelange, 2018-12-22:
@SvenRoederer is there a reason that we haven't merged this into master yet?