freifunk-berlin/firmware #514: now TARGET: ar71xx-tiny

SvenRoederer, 2018-01-21:
this was added in OpenWRT with commit https://github.com/openwrt/openwrt/commit/0cd5e85e7ad621223b0787e66d8ad20fb2694135.
It's a special target (more size-optimized) for 4MB devices.

this new TARGET needs to be adapted into Firmware and buildbots

SvenRoederer, 2018-03-12:
Or should we take this as chance to remove the 4MB devices ?

torte71, 2018-03-12:
- Being a new target, this means an increased size of ~500MB for each build on the buildbot.
- The "backbone" flavour can be built, but is too big to be useable (too little space for config partition).
- "default" is useable at this point. But will it still be, if any of the contained packages grows?

My preference: Drop official support for 4mb devices in the next major release.
We may keep the files for building the "default" flavour, but don't provide readily built images. So people, that really want to stick to those devices can do so - if they are willing to invest the time for building, etc. .

booo, 2018-08-21:
As far as I can see we introduced the ar71xx-tiny with commit 36e8bf0519f6db49a2921f872e2bf2277943efb7. 

Can we close the ticket or is there still something to discuss? I suggest that we discuss general support of 4mb devices in another ticket.

bobster-galore, 2018-09-21:
It could work with tiny-tc?
@torte71 what do u think?
If there is no more heartbeat with this we can close it.

closed 2018-09-30 by bobster-galore