freifunk-berlin/firmware #515: CONFIG-Option "VERSION_NICK" has been dropped upstream

SvenRoederer, 2018-01-22:
In https://github.com/openwrt/openwrt/commit/d573b1287e1b4ace7680d5dbfde20ba25e428b17 wirde in OpenWRT die CONFIG-Option "VERSION_NICK" abgeschafft. In diesem Feld haben wir bisher immer unseren Releasename ("Kathleen", "Hedy", ...) gehabt.
Ich sehe folgende Möglichkeiten:
* wir verzichten drauf, dass der Releasename in der Firmware auftaucht und haben dann einfach nur die Versionsnummern
* wir legen die Parameter "VERSION_DIST" und "VERSION_NICK" zusammen ("VERSION_DIST=Freifunk Berlin Hedy")

SvenRoederer, 2018-01-22:
lt. Commit https://github.com/openwrt/openwrt/commit/d23e1e1e1a3eede59d7da21cafe0ba69c0e81458#diff-973cc0b123811266b40567a4554dd8a6R44 kann es sein, dass die Leerzeichen, die wir nutzen das "opkg" verwirren.
Mit https://github.com/freifunk-berlin/firmware/commit/a1d918c69291deb9f2117cdd69813837358582cb probier ich das mal aus.

bobster-galore, 2018-01-22:
We should consider that it doesn't break the here https://github.com/freifunk-berlin/firmware/issues/22#issuecomment-357826321 mentioned autoupdater ( [app](https://github.com/openwrt/packages/tree/master/utils/auc) ).

bobster-galore, 2018-09-21:
And what was the outcome of ur test?
Is it still valid? Is it a blocker?

bobster-galore, 2018-09-30:
As proposed by @SvenRoederer we combine to ("VERSION_DIST=Freifunk_Berlin_Hedy") and use underscores to circumvent problems which probably arise from whitespaces?

SvenRoederer, 2018-11-13:
Please comment my idea from https://github.com/freifunk-berlin/firmware/commit/359e7ff2f5ea8d8ceb6bcea9b2f1740f043a9e6d.

As the CONFIG-Values directly map to `/etc/openwrt_release`
> DISTRIB_ID='Freifunk Berlin' <-- CONFIG_VERSION_DIST
DISTRIB_RELEASE='2018-alpha' <-- CONFIG_VERSION_NUMBER
DISTRIB_REVISION='6e8f9f1' <-- git-version / git-tag

this ensues we have the Release only in one datafield (DISTRIB_RELEASE) and not spread over ID and RELEASE, which will make parsing more painful

SvenRoederer, 2018-12-12:
Any progress in deciding if we will go like 0a16b523d46a6d9febef5c17daf91c675ab1953f or 
6bda7f7 ?

SvenRoederer, 2019-03-02:
fixed by 38d417268b53db0803038f36ac4ef4bc26f219de + f8847ffb682a6f0c615f6afd9624ca8ae16f2969

closed 2019-03-02 by SvenRoederer