freifunk-berlin/firmware #516: Project Fi

pmelange, 2018-01-23:
There is a new way that google wants to break the internet.  It's called Project Fi.  They want to make it so that in Android, the OS can automatically tell you how fast local open wireless networks are before joining them.  There are a few ways to opt out.

https://support.google.com/fi/answer/6188557

The Canary URL seems like to best option.

SvenRoederer, 2018-01-23:
What's the problem with "Project FI"?
According to PicoPeeringAgreement Freifunk will not filter or block traffic.

kehugter, 2018-03-10:
> What's the problem with "Project FI"?

AFAIU, Google abuses open Wi-Fi networks to provide their paid service.

> Freifunk will not filter or block traffic

The Canary URL will prevent automated abuse, not legit traffic. It doesn't go against the Pico Peering Agreement IMO.

booo, 2018-08-21:
The problem is part of the android operating systems. Android users should fight these configurations and implementations. The freifunk firmware is the wrong place to fight these problems.

Apart from this we are an open network and should not restrict the use of our network even if the user is a nasty bot net of android devices.

booo, 2018-08-21:
I would like to close this ticket if there are no objections.

closed 2018-09-15 by booo