freifunk-berlin/firmware #517: Wifi Signalstärke auf auto

carsifreifunk, 2018-01-23:
Am Sonntag habe ich auf einem WDR3600 Hedy-1.0.0-rc1 (müsste build 595 gewesen sein) installiert und festgestellt, dass die Sendeleistung auf "auto" unter Wifisettings stand und nur mit 50% Sendeleistung daher kam. Ich glaube nicht, dass das gewünscht ist.

SvenRoederer, 2018-01-23:
kannst du mal bitte die "/etc/config/wireless" direkt nach dem Zurücksetzen auf Werkseinstellungen, auch ohne den Assistenten laufen zu lassen, zeigen?

SvenRoederer, 2018-01-23:
Ich hab grad mal meinen CPE210 neugemacht (revision ab3754cb311c84fb99adc61ab583ec0198510ecc) und kann im LuCI auch sehen, dass die Leistung auf "auto" steht.
wenn ich auf der console mit "iwinfo" nachsehe, habe ich aber 20 dBm.
```
root@ff-tk-spare2:~# iwinfo 
wlan0-adhoc-2 ESSID: "intern-ch13.freifunk.net"
          Access Point: D2:CA:FF:EE:BA:BE
          Mode: Ad-Hoc  Channel: 13 (2.472 GHz)
          Tx-Power: 20 dBm  Link Quality: 40/70
          Signal: -70 dBm  Noise: -89 dBm
          Bit Rate: 12.1 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan0-dhcp-2 ESSID: "berlin.freifunk.net"
          Access Point: 86:16:F9:88:70:DE
          Mode: Master  Channel: 13 (2.472 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -89 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

SvenRoederer, 2018-01-23:
@carsifreifunk war es wirklich build 595 für ar71xx?
Das ist dann OpenWRT-master aka "Hedy-1.1.0".

carsifreifunk, 2018-01-24:
Ich habe jetzt Hedy 1.0.0-rc1 d598dd7 drauf und nach durchlauf des wizards (für Potsdam) ist die Sendeleistung unter Wifi wieder auf auto. 

Auszug aus /etc/config/wireless

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/ar934x_wmac'
	option htmode 'HT20'
	option disabled '0'
	option country 'DE'
	option doth '0'
	option diversity '1'
	option channel '5'
	option hwmode '11g'

config wifi-iface
	option network 'wireless0'
	option encryption 'none'
	option device 'radio0'
	option bssid '02:CA:FF:EE:BA:BE'
	option mcast_rate '6000'
	option mode 'adhoc'
	option ifname 'wlan0-adhoc-2'
	option ssid 'Mesh23'

config wifi-iface
	option ifname 'wlan0-dhcp-2'
	option network 'dhcp'
	option encryption 'none'
	option device 'radio0'
	option mode 'ap'
	option ssid 'Testing-Hedy'

SvenRoederer, 2018-01-25:
@carsifreifunk kannst du mal prüfen, wie in Kommentar #3 (https://github.com/freifunk-berlin/firmware/issues/517#issuecomment-359940569) von mir gemacht?

carsifreifunk, 2018-01-27:
In iwinfo stehen auch bei mir 20db
wlan0-adhoc-2 ESSID: "Mesh23"
          Access Point: 02:CA:FF:EE:BA:BE
          Mode: Ad-Hoc  Channel: 5 (2.432 GHz)
          Tx-Power: 20 dBm  Link Quality: 36/70
          Signal: -74 dBm  Noise: -95 dBm
          Bit Rate: 6.2 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

Wenn ich jetzt in olsr die Verbindung zum direkten Nachbar anschaue, ist diese jetzt auch auf 1. Bei einem anderen Build war da ein schlechterer Wert. Unter Wifi in LuCi wird jetzt 1005 Signalstärke zu diesem Nachbar angezeigt.

pmelange, 2018-11-04:
I am closing this issue due to inactivity.  If you still have a problem, please reopen

closed 2018-11-04 by pmelange