freifunk-berlin/firmware #522: olsr watchdog does not delete old smartgateway tunnel

pmelange, 2018-02-08:
When the olsr watchdog notices that the olsr daemon no longer runs, it simply restarts the daemon.  But the smartgateway tunnel "tnl_+" is not deleted.  When the olsr daemon then restarts, it is unable to create a tunnel if the tunnel to the same destination already exists.  See the error log below.

```
Thu Feb  8 17:55:00 2018 user.notice OLSR watchdog: Process died - restarting!
Thu Feb  8 17:55:15 2018 daemon.info olsrd[2162]: Tunnel tnl_0ae6f052 added, to 10.230.240.82
Thu Feb  8 17:55:16 2018 daemon.info olsrd[2162]: Tunnel tnl_0ae6f052 removed, to -
Thu Feb  8 17:55:16 2018 daemon.info olsrd[2162]: Tunnel tnl_4d573032 added, to 77.87.48.50
Thu Feb  8 17:55:41 2018 daemon.info olsrd[2162]: Tunnel tnl_4d573032 removed, to -
Thu Feb  8 17:55:41 2018 daemon.err olsrd[2162]: Cannot add tunnel tnl_0a1f0b01 to 10.31.11.1: File exists (17)
Thu Feb  8 17:55:41 2018 daemon.err olsrd[2162]: Cannot create tunnel tnl_0a1f0b01
```
By deleting the tunnel and restarting olsr, everything works again.

SvenRoederer, 2018-02-09:
Sounds  to me like only a cosmetically issue.
When the tunnel-setup fails, as the exact same tunnel is already setup, this will cause the an error in the setup-function. But as the tunnel is still correctly configured I don's see any functional problem arising from this.

pmelange, 2018-02-10:
I disagree.  I believe that when the smartgateway plugin is not able to create the tunnel (because the tunnel already exists) that it also does not set up the necessary firewall rules.  I believe it can lead to unexpected behaviour.

Another case example is if the old existing tunnel interface goes to a node which no longer exists, then olsrd restarts and builds a new tunnel interface.  But in the routing table, the old tunnel interface is still there.  This can also lead to unexpected behaviour.

pmelange, 2018-06-13:
A neighbor node of mine recently had an issue with this.  No more internet was available because after each olsrd crash, the tunnels weren't removed.  

When I logged in, there were 7 stale tunnels, and the olsr daemon couldn't create a new one (unable to bind to link name since it already existed).  I had to manually delete each tunnel instance and restart olsrd for the node to start working again.

I highly recommend that the changes in pr #1761 in openwrt/luci be included in the next Hedy release.

pmelange, 2018-07-20:
The pull request with openwrt/luci has been sitting idle for 3 months.  Should I submit this as a patch for freifunk-berlin only?  It would be unfortunate because this change could be useful to other communities as well.

pmelange, 2018-07-26:
merged in openwrt/luci#1761

closed

closed 2018-07-26 by pmelange