freifunk-berlin/firmware #523: Community Profile unter Hedy veraltet/anders

carsifreifunk, 2018-02-19:
Es scheint, dass das die Quelle für die communityprofile öfter Wechseln. Unter Kathleen hat er sich wohl noch https://git.openwrt.org/?p=project/luci.git;a=blob_plain;f=contrib/package/community-profiles/files/etc/config/profile_potsdam;hb=HEAD gezogen. Unter Hedy ist davon nichts mehr da.

SvenRoederer, 2018-02-19:
kathleen0.3.0 basiert auf OpenWRT 15.05, folglich haben wir auch den Luci-branch "for15.05" genutzt (https://github.com/openwrt/luci/commit/d89b022b97b4516bef708212135780b818b9d4a2) . Dieser Branch wurde irgendwann mal vom "LuCI-master" abgespalten.
Für Hedy-1.0.0 nutzen wir folglich den LuCI-branch "lede-17.01" (https://github.com/openwrt/luci/commit/3d46f2593a821960f896e109e1d6f89c481d6004). Dieser wurde auch irgendwann vom "LuCI-master" abgespalten, aber wesetlich später als der "for-15.05"-branch.

Wenn jetzt eine Änderung in den master-branch gemacht wird, die aber nicht in den release-branch "lede-17.01" geportet wird, entsteht hier eine differenz.

> sven@build:/mnt/hosts/build/src/openwrt/freifunk/lede/openwrt/feeds/luci$ git diff d89b022b97b4516bef708212135780b818b9d4a2..3d46f2593a821960f896e109e1d6f89c481d6004 -- contrib/package/community-profiles/files/etc/config/profile_potsdam 

zeigt keine Unterschiede

~~~
sven@build:/mnt/hosts/build/src/openwrt/freifunk/lede/openwrt/feeds/luci$ git diff d89b022b97b4516bef708212135780b818b9d4a2..origin/master -- contrib/package/community-profiles/files/etc/config/profile_potsdam 
--- a/contrib/package/community-profiles/files/etc/config/profile_potsdam
+++ b/contrib/package/community-profiles/files/etc/config/profile_potsdam
@@ -1,19 +1,35 @@
 config 'community' 'profile'
        option 'name' 'Freifunk Potsdam'
        option 'homepage' 'http://potsdam.freifunk.net'
-       option 'ssid' 'www.freifunk-potsdam.de'
+       option 'ssid' 'Freifunk-Potsdam-XXX-YYY'
        option 'mesh_network' '10.22.0.0/16'
        option 'splash_network' '192.168.22.0/24'
        option 'splash_prefix' '24'
        option 'latitude' '52.39349'
        option 'longitude' '13.06489'
+       option 'ipv6' '0'
+
+config 'defaults' 'interface'
+       option 'netmask' '255.255.0.0'
+       option 'dns' '85.214.20.141 213.73.91.35 194.150.168.168'
+       option 'delegate' '0'
 
 config 'defaults' 'wifi_device'
        option 'channel' '5'
 
+config 'defaults' 'wifi_device_5'
+       option 'channel' '44'
+
 config 'defaults' 'bssidscheme'
        option '5' '02:CA:FF:EE:BA:BE'
+       option '44' '02:CA:FF:EE:BA:BE'
 
-config 'defaults' 'interface'
-        option 'dns' '85.214.20.141 213.73.91.35 194.150.168.168'
+config 'defaults' 'ssidscheme'
+       option '5' 'Mesh23'
+       option '44' 'Mesh23'
+
+config 'defaults' 'dhcp'
+       option 'leasetime' '15m'
 
+config 'defaults' 'olsrd'
+       option 'LinkQualityAlgorithm' 'etx_ffeth'
~~~

niccokunzmann, 2018-02-20:
> Für Hedy-1.0.0 nutzen wir folglich den LuCI-branch "lede-17.01" (openwrt/luci@3d46f25).

Ich schlussfolgere: Wenn ein Community-Profil in dem  lede-17.01-Branch ist, dann kommt es in den Hedy-Build.

[In der Contributing.md](https://git.openwrt.org/?p=project/luci.git;a=blob;f=CONTRIBUTING.md;h=e4edf0c1519ebe18d2fab0efe517e0afb864ef31;hb=refs/heads/master) kann ich folgendes lesen:

> - Please ONLY cherry-pick or commit security and bug-fixes to these branches.
> - If you are unsure if your change is suitable, please use a pull request.

Ich denke, dass wir die Commits vom Master-Branch cherry-picken können und als Pull-Request and den lede-17.01 branch stellen können.

Links:
- [Branches](https://git.openwrt.org/?p=project/luci.git;a=heads)
- [Github Repository](https://github.com/openwrt/luci)
- [Potsdam Profil im Repo, master branch](https://github.com/openwrt/luci/blob/master/contrib/package/community-profiles/files/etc/config/profile_potsdam)

@carsifreifunk Ist das [Profil](https://github.com/openwrt/luci/blob/master/contrib/package/community-profiles/files/etc/config/profile_potsdam) so aktuell und richtig?

Die verschiedenen Historien:
- [master](https://github.com/openwrt/luci/commits/master/contrib/package/community-profiles/files/etc/config/profile_potsdam)
- [lede-17.01](https://github.com/openwrt/luci/commits/lede-17.01/contrib/package/community-profiles/files/etc/config/profile_potsdam)

Commits, die wir brauchen:
- https://github.com/openwrt/luci/commit/892398692c99389f8a0e3b44f1054a639f010257#diff-9a74912dd9f9c141a75d5ff66ae0b2ea

niccokunzmann, 2018-02-20:
Weiter gehts mit:
- [ ] [Pull request](https://github.com/openwrt/luci/pull/1634)
- [ ] Sollte diese Pull-Request in den lede-17.01-Branch eingepflegt werden, entsteht dort ein neuer Commit, der im Hedy-Build branch als der neue Commit für den Build ausgecheckt werden muss, denke ich.

SvenRoederer, 2018-02-20:
JA, soweit alles korrekt.

Alternativ kann das profile auch via patch im firmware-berlin repo geändert werden. Es kann sich u.U. ja etwas hinziehen, bis der PR bearbeitet wird und als Patch ist er dann schonmal im aktuellen Hedy-build.

SvenRoederer, 2018-02-20:
ich hab mir mal die aktuelle Firmware angesehen und beabsichtige zum Ende des Wochenendes dem Code den Namen "v1.0.0" zu geben.

Bis dahin sollte euer Profile-update
* im luci-branch geändert sein
* ihr eine Patch zum Mergen (PR) schicken (ala https://github.com/freifunk-berlin/firmware/blob/340a94cf52bdaf20f9606c1272cffc5236f91d67/patches/200-update-community-profiles.patch)
* oder auf das nächste release warten

niccokunzmann, 2018-02-24:
@SvenRoederer Da die nun nicht reagieren, an welchen Branch würdest du gern den Patch haben?
Es gibt mehrere Hedy-branches.

SvenRoederer, 2018-02-24:
nehmt den "master" , dann rebase ich das dann weiter in den "Hedy-1.0.0"

niccokunzmann, 2018-02-25:
@carsifreifunk @SvenRoederer Der Commit wurde cherry-picked. Weiter mit Schritt 2: Ein Update des master branches.

carsifreifunk, 2018-02-26:
@niccokunzmann sieht alles gut aus und frue mich, dass es voran geht.
Steht jetzt noch
option 'ssid' 'www.freifunk-potsdam.de'
drin?
Wir hatten uns ja auf 
option 'ssid' 'freifunk-potsdam.de'
geeinigt.

SvenRoederer, 2018-02-26:
aktuell ist "https://github.com/openwrt/luci/blob/c1e6a3dd3ae1be749c8608ffec8d9c90362e17db/contrib/package/community-profiles/files/etc/config/profile_potsdam#L4", wie im master-branch.

Wenn noch was anderes rein soll, dann schnell ein patch gegen unser Repo. Der Code-freeze ist eigentlich schon durch. Und ich baue grad die Final-images ... (http://paco.geroedel.de/openwrt/freifunk-berlin/Hedy-1.0.0/)

niccokunzmann, 2018-02-26:
@carsifreifunk, Für mich ist das eine neue Issue. Diese hier ist dazu da, dass das alte Profil mit reingenommen wird. 
Ich schlage vor, um die Diskussion nachvollziehbar zu halten und neue Diskutanten mit reinzubekommen, erstelle eine neue Issue, dass das Profil geändert werden soll und wir diskutieren, was gemacht werden soll.
Wenn diese fertig ist, können wir sie schließen.

hafu, 2018-03-01:
Kann geschlossen werden. Ist im maser und lede-17.01 Branch. Siehe
- openwrt/luci#1650
- openwrt/luci#1651

niccokunzmann, 2018-03-01:
@carsifreifunk Wir können sie nicht schließen. Kannst du sie schließen?

closed 2018-03-02 by SvenRoederer