freifunk-berlin/firmware #531: Wizard: Leaving the "Mesh-IP" entry empty runs into lua error

torte71, 2018-03-11:
Pressing "Save and reboot" with an empty "Mesh-IP" entry runs into following lua error:
```
/usr/lib/lua/luci/dispatcher.lua:380: Failed to execute cbi dispatcher target for entry '/admin/freifunk/assistent/wireless'.
The called action terminated with an exception:
...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:49: bad argument #1 to 'IPv4' (string expected, got nil)
stack traceback:
        [C]: in function 'assert'
        /usr/lib/lua/luci/dispatcher.lua:380: in function 'dispatch'
        /usr/lib/lua/luci/dispatcher.lua:109: in function </usr/lib/lua/luci/dispatcher.lua:108>
```

SvenRoederer, 2018-10-16:
fixed by https://github.com/freifunk-berlin/firmware-packages/commit/2bcd91fdd0601f133e8931e7b3a1f92ae6df28f1

SvenRoederer, 2018-10-16:
we should recheck what happens when entering no data into the field.

bobster-galore, 2018-10-17:
May be @torte71 could check cause he knows the situation best?

torte71, 2018-12-21:
Checked with following versions (always the tunneldigger image for cpe210v1):
- Branch "alpha-test-tunneldigger", revision 09984f8 compiled from source: Works correctly
- Branch "master", revision 1f030e9 from buildbot directory 1062: Works correctly
- Branch "Hedy-1.0.2", revision 52c60b6 from buildbot directory 1063: Still broken (patch not applied, afaics)

pmelange, 2018-12-21:
@torte71 I have made PR https://github.com/freifunk-berlin/firmware-packages/pull/176 to have this code merged.

SvenRoederer, 2019-02-16:
added to master in 2bcd91fdd0601f133e8931e7b3a1f92ae6df28f1 and backported for Hedy-1.0.2, too

closed 2019-02-16 by SvenRoederer