freifunk-berlin/firmware #532: no-tunnel/default: Can't ping gateway, totally disconnected after ffwizard

vasyugan, 2018-03-14:
I downloaded the default/no-tunnel flavour in order to avoid the VPN bottleneck. After the initial Freifunk setup, the router is totally disconnected, I can't even ping the default gateway on the lan. Experienced with a TP-Link CPE 210 and a GL.iNet GL-AR150. It did work with a Ubitquiti Nanostation.

I compared the etc/config/firewall of the nanostation with the others and they seem identical. Could this be related to VLAN issues? 

I will gladly help debugging this further if given instructions.

SvenRoederer, 2018-03-16:
see also the MAilinglis at https://lists.berlin.freifunk.net/pipermail/berlin/2018-March/037272.html

vasyugan, 2018-03-16:
Yes, that's me.

vasyugan, 2018-03-16:
Just tried again today, with a CPE 210, same shitty result. both with the setup resulting from the FFWizard and reboot and from restoring an earlier config from Kathleen 0.3.

vasyugan, 2018-03-17:
So now, after a number of tries have narrowed it in so far that the problem occurs after I run the FF wizard _with Internet sharing_ while  running the wizard to merely join the Freifunk network works. So what settings does the FF wizard change in the former case?

SvenRoederer, 2018-04-01:
On the mailinglist (https://lists.berlin.freifunk.net/pipermail/berlin/2018-March/037323.html) was commented what the wizard is doing.
Did it help?

You also mentioned something about VLANs (https://lists.berlin.freifunk.net/pipermail/berlin/2018-March/037355.html), probably that's related to your issue

torte71, 2018-04-08:
Could this be related to this: https://github.com/freifunk-berlin/firmware/issues/545?
In that case, entering `/etc/init.d/network restart` on the router's shell should give connectivity.

Another idea: Are the FF-routers directly connected to your home router, or is there some hardware between, e.g. wifi-repeaters, switches? This may have something to do with the dynamic MAC-address of the ffuplink interface.

everloop2, 2018-04-12:
maybe this works: https://github.com/freifunk-berlin/firmware/issues/542#issuecomment-380450529

SvenRoederer, 2018-07-25:
probably fixed by https://github.com/freifunk-berlin/firmware-packages/commit/ec75de63ef18d88cafe74366e39257b824cd8818

feel free to reopen if it still fails

closed 2018-07-25 by SvenRoederer