freifunk-berlin/firmware #533: olsrd nullpointer

PolynomialDivision, 2018-03-16:
Hi,
I was using 625 unstable build.
There was a [invalid write access](https://gist.github.com/PolynomialDivision/77280b5916a42552e93a53b90883bfbe#file-null_olsrd) at startup.

    Fri Mar 16 15:29:04 2018 kern.info kernel: [   92.880783] 
    Fri Mar 16 15:29:04 2018 kern.info kernel: [   92.880783] do_page_fault(): sending SIGSEGV to olsrd for invalid     write access to 004687f0
    Fri Mar 16 15:29:04 2018 kern.info kernel: [   92.889205] epc = 00417249 in olsrd[400000+38000]
    Fri Mar 16 15:29:04 2018 kern.info kernel: [   92.894013] ra  = 0041dc4f in olsrd[400000+38000]
    Fri Mar 16 15:29:04 2018 kern.info kernel: [   92.898794] 

hedy-1.0.0-8ff420b-archer-c7-v2-sysupgrade.bin

torte71, 2018-03-16:
Does "master" use olsrd-0.9.0.3 as well?
Then it's a duplicate of https://github.com/freifunk-berlin/firmware/issues/513

SvenRoederer, 2018-03-16:
yes, it does (https://github.com/freifunk-berlin/firmware/blob/8ff420b0ad264dddb48b29f890055f98498f3829/patches/300-use_olsrd0903.patch)

closed 2018-03-16 by SvenRoederer