freifunk-berlin/firmware #535: Hedy-1.0.1 release discussion

sarumpaet, 2018-03-16:
To have a process that is a bit more transparent than the last release (Hedy-1.0.0), we should coordinate the Hedy-1.0.1 release here and release it only once several developers agreed, and all [Hedy-1.0.1 blocker issues](https://github.com/freifunk-berlin/firmware/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+milestone%3AHedy-1.0.1+label%3Ablocker) are resolved.

SvenRoederer, 2018-03-16:
As long as some developers are interested in working on a release, this might work.

bobster-galore, 2018-03-17:
>'''To  have a process that is a bit more transparent than the last release  (Hedy-1.0.0), we should coordinate the Hedy-1.0.1 release here and  release it only once several developers agreed, and all Hedy-1.0.1 blocker issues are resolved.

To me it seemed transparent as it was already announced on the berlin freifunk ml. Since there was no reaction vetoing a release, to me nothing looked like being wrong.

What does a blocker help, when only very few ppl are active in developing a new version? Getting more time to wait for a new release? 
I like coordinated work as well but it would be nice if some are coworking.

sarumpaet, 2018-04-23:
We had issues like this before, see https://github.com/freifunk-berlin/firmware/issues/367 , and this worked fine for coordination, and it increased transparency and communication between developers. We also had hackathons previously which were good for teamwork. I set up a Doodle - perhaps we manage to get together again this time (link see list).

SvenRoederer, 2018-05-02:
I think the branch Hedy-1.0.1 is good to provide the migration from 0.3.0 -> 1.0.1 and 1.0.0 -> 1.0.1.
The individual software-versions are not changed and the same as in 1.0.0 release.

I called the mailinglist for testers (https://lists.berlin.freifunk.net/pipermail/berlin/2018-May/037667.html)

sarumpaet, 2018-05-22:
I think https://github.com/freifunk-berlin/firmware/pull/566 should be cherry-picked to 1.0.1 as well since it fixes a regression.

sarumpaet, 2018-05-29:
Re https://github.com/freifunk-berlin/firmware/pull/566 (not) in 1.0.1 - I'd just release fixes as they become available but whatever. 1.0.1 go! I created https://github.com/freifunk-berlin/firmware/pull/568 for 1.0.2.