freifunk-berlin/firmware #536: Wechsel auf 802.11s als MESH-Protokoll für Layer1

SvenRoederer, 2018-03-19:
Wir sollten das Netz auf 802.11s meshing umstellen.

* teilweise haben WiFI-Chips keinen funktionierenden Support für AP und AdHoc-mode gleichzeitig (Archer C7v2 #346, GiInet mt300n (mit OpenWRT master), bestimmt auch TPlink WR841v13)
* kurze tests haben ergeben, dass die 802.11s-Verbindungen "flüssiger" reagieren als die entsprechende AdHoc-Version

SvenRoederer, 2018-03-19:
1. Machen wir das als DEfault für das nächste Major-Release?
1. wie machen wir die Umstellung vorhandener Router
   1. automatisch würde die Umstellung des vorhandenen AdHoc-interfaces funktionieren
   1. für parallelbetrieb von altem AdHoc- und neuem Mesh-interface muss User neue IP-Adressen beschaffen und eingeben

pmelange, 2018-03-23:
Answering SvenRoederer:
1. I would really like to see it as a default.
2. ii. Is it possible to have the wireless0 device be a bridge with both an adhoc and 80211s wireless interface, using just one IP addr?

SvenRoederer, 2018-03-23:
> Is it possible to have the wireless0 device be a bridge with both an adhoc and 80211s wireless interface, using just one IP addr?

* there is a 50% chance that you can bridge an Adhoc interface
* just thinking of 2 nodes having a bridge of Adhoc and mesh, and both see each other. This will create a loop and might confuse the routing-deamons

bobster-galore, 2018-03-30:
There's a project here https://github.com/freifunk-berlin/firmware/projects/3

nExXxuS, 2018-04-12:
My propsal would be a fixed date, when we will change the mesh protocol. The change then happens everywhere simultanously using a cronjob, which will do the transition + network reload when that date has come. The cronjob could be distributed with hedy 1.1.

nExXxuS, 2018-04-13:
We should also consider to provide a script which runs on boot - and checks wether that date is already over - and apply the changes then once. Otherwise routers which are powered down during that time would not get the config change.

bobster-galore, 2018-05-30:
I propose we call (mesh_id) the mesh network "mesh.berlin.freifunk", which would be a describing name for what it does. Furthermore it makes clear which settings need to be met for a functioning connection as they differ from community to community.

bobster-galore, 2018-06-26:
At recent Spandau meeting we set up mesh (11s) with mesh id "mesh-berlin-freifunk" which was considered to be a nice idea as it contains the community and doesn't read like an internet address.

booo, 2018-08-21:
@bobster-galore could you please post a working default configuration?

I suggestion that we implement the change to the wizard in the near future and discuss the actual transition in the next step, e.g. in one of our meetings.

SvenRoederer, 2018-11-19:
There is a PR open for doing 802.11s by default: https://github.com/freifunk-berlin/firmware-packages/pull/156