freifunk-berlin/firmware #540: OLSR crashes but leaves an unkillable zombie process

pmelange, 2018-04-01:
I have had the following happen to me three time since installing the firmware on an ERX-SFP (Hedy 1.0.0).

OLSR crashes and there is nothing I can do to restart it.  If I run the neigh.sh script, it hangs.  If I run /etc/init.d/olsrd restart, it hangs.  If I kill -9 the PID, the process stays in the ps list.

The folowing kernel message is shown every 5 to 10 seconds:
```
kern.emerg kernel: [1054204.339802] unregister_netdevice: waiting for tnl_010b1f0a to become free. Usage count = 1
```

And when the OLSR watchdog runs, I get:
```
cron.info crond[1532]: USER root pid 26254 cmd /usr/sbin/ff_olsr_watchdog
daemon.err olsrd: /etc/init.d/olsrd: ERROR: there is already an instance of olsrd running (pid: '10264'), not starting.

Since killing the process doesn't work, the only other alternative is to reboot the device.

pmelange, 2018-04-01:
There is a clear change in memory usage too.

[image]

SvenRoederer, 2018-04-01:
as mentioned by the olsrd-maintainers, we should upgrade to olsr 0.9.6.1+ (https://github.com/freifunk-berlin/firmware/issues/424#issuecomment-287652084)

bobster-galore, 2018-04-10:
... and u can't kill a zombie since it's already dead!
U might kill everything above and around it, to let the zombie rest in piece.

pmelange, 2018-04-24:
This has happened to me 4 more times since I wrote this issue.  I am going to replace the ERX-SFP with a WDR4900 to see if the problem persists.  

@bobster-galore silly

pmelange, 2018-07-18:
This was only a problem with the ERX-SFP

closed 2018-07-18 by pmelange

SvenRoederer, 2018-07-19:
what caused this on your ERX?

pmelange, 2018-07-20:
Unknown.  The ERX is not longer being used with openwrt.  All the info I had I put in the original comment