freifunk-berlin/firmware #542: no-tunnel/default: ERR_CONNECTION_TIMED_OUT

everloop2, 2018-04-02:
git: Hedy-1.0.1
target: lantiq (EasyBox803), maybe others?
firmware: no-tunnel (default)

ffwizard completed (share internet, monitoring, setting IPs..),
WAN set from nas0 to eth0.x (1 of 4 LAN ports),
WAN in bridge mode: eth0.x and ffuplink_wan
FFUPLINK gets IP (from cable modem)

> trying to connect to a WWW-Page > ERR_CONNECTION_TIMED_OUT
> LuCI > Network > Diagnostics > ping, traceroute, nslookup **working**

**workaround**: setting up an other "private DHCP" on eth0.y in its own firewall zone allowing forward to "ffuplink zone" > bug is not VLan related

changing firewall, interfaces, routing table settings did not help

ff-tunnel interface is not needed anymore (notunnel/default firmware)?
can this be freifunk-policyrouting related?

SvenRoederer, 2018-04-03:
what is the ```ff-tunnel interface``` you mentioned?

everloop2, 2018-04-03:
maybe its related to: https://github.com/freifunk-berlin/firmware/issues/532, tried hints given there, had no luck.

everloop2, 2018-04-03:
Freifunk: DHCP (eth0.1 + Master "cottbus.freifunk.net")
WAN: DHCP-client (eth0.2 + ffuplink_wan) 
private: DHCP (eth0.3 + Master "private")

```
root@cottbus-lausi36:~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 74:31:70:2b:b8:fc brd ff:ff:ff:ff:ff:ff
    inet6 fe80::7631:70ff:fe2b:b8fc/64 scope link 
       valid_lft forever preferred_lft forever
3: tunl0@NONE: <NOARP,UP,LOWER_UP> mtu 1480 qdisc noqueue state UNKNOWN group default qlen 1
    link/ipip 0.0.0.0 brd 0.0.0.0
7: ifb0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc hfsc state UNKNOWN group default qlen 32
    link/ether 4e:c0:f6:03:c6:3b brd ff:ff:ff:ff:ff:ff
    inet6 fe80::4cc0:f6ff:fe03:c63b/64 scope link 
       valid_lft forever preferred_lft forever
8: br-dhcp: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 74:31:70:2b:b8:fc brd ff:ff:ff:ff:ff:ff
    inet 10.35.0.97/27 brd 10.35.0.127 scope global br-dhcp
       valid_lft forever preferred_lft forever
    inet6 fd42:8598:4def::1/64 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::7631:70ff:fe2b:b8fc/64 scope link 
       valid_lft forever preferred_lft forever
9: eth0.1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-dhcp state UP group default qlen 1000
    link/ether 74:31:70:2b:b8:fc brd ff:ff:ff:ff:ff:ff
10: ffuplink_wan@ffuplink: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-wan state UP group default qlen 1000
    link/ether 7a:9d:19:a2:b7:89 brd ff:ff:ff:ff:ff:ff
11: ffuplink@ffuplink_wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc hfsc state UP group default qlen 1000
    link/ether c6:34:72:c5:82:d8 brd ff:ff:ff:ff:ff:ff
    inet 172.30.179.156/17 brd 172.30.255.255 scope global ffuplink
       valid_lft forever preferred_lft forever
    inet6 fe80::c434:72ff:fec5:82d8/64 scope link 
       valid_lft forever preferred_lft forever
12: br-wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 7a:9d:19:a2:b7:89 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::789d:19ff:fea2:b789/64 scope link 
       valid_lft forever preferred_lft forever
16: eth0.2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-wan state UP group default qlen 1000
    link/ether 74:31:70:2b:b8:fc brd ff:ff:ff:ff:ff:ff
51: br-local_dhcp: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 74:31:70:2b:b8:fc brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-local_dhcp
       valid_lft forever preferred_lft forever
    inet6 fe80::7631:70ff:fe2b:b8fc/64 scope link 
       valid_lft forever preferred_lft forever
52: eth0.3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-local_dhcp state UP group default qlen 1000
    link/ether 74:31:70:2b:b8:fc brd ff:ff:ff:ff:ff:ff
53: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-local_dhcp state UP group default qlen 1000
    link/ether 74:31:70:2b:b8:fc brd ff:ff:ff:ff:ff:ff
    inet6 fe80::7631:70ff:fe2b:b8fc/64 scope link 
       valid_lft forever preferred_lft forever

```

```
root@cottbus-lausi36:~# ip rule
0:	from all lookup local 
1000:	from all lookup olsr 
2000:	from all lookup localnets 
19989:	from all to 172.30.128.0/17 iif tunl0 prohibit
19989:	from all to 172.30.128.0/17 iif prohibit [detached] lookup main 
19989:	from all to 172.30.128.0/17 iif br-dhcp prohibit
19990:	from all iif tunl0 lookup ffuplink 
19990:	from all iif br-dhcp lookup ffuplink 
19999:	from all iif wlan0-adhoc-2 [detached] lookup olsr-tunnel 
19999:	from all iif tunl0 lookup olsr-tunnel 
19999:	from all iif br-dhcp lookup olsr-tunnel 
20000:	from all iif wlan0-adhoc-2 [detached] lookup olsr-default 
20000:	from all iif tunl0 lookup olsr-default 
20000:	from all iif br-dhcp lookup olsr-default 
20001:	from all iif wlan0-adhoc-2 [detached] unreachable
20001:	from all iif tunl0 unreachable
20001:	from all iif br-dhcp unreachable
32766:	from all lookup main 
32767:	from all lookup default 
100000:	from all lookup olsr-tunnel 
100010:	from all lookup olsr-default 
root@cottbus-lausi36:~# 
```

everloop2, 2018-04-11:
tried https://github.com/freifunk-berlin/firmware/issues/545#issuecomment-379536617 "/etc/init.d/network" restart after box is up, did not work

everloop2, 2018-04-11:
got it workin > did not know that **WAN and FFUPLINK both needs IP assigend** > "DHCP mode"

on EasyBox before/after ff-wizard WAN is in "pppoe mode" > DSL uplink
-> have cable modem -> set a LAN port via vlan as uplink

before tried an other config working only for a seperate "LOCAL-DHCP":
WAN (unmanaged) and FFUPLINK (dhcp mode), freifunk-firewall-zone had no connection and local-dhcp-firewall-zone had connection via FFUPLINK -> this is where i stuck

Is it essential to have 2 uplink IPs?

everloop2, 2018-04-11:
did work for 5min > connection loss, done Box reset and ff-wizard again,
if "WAN cable modem" behind Easybox is resetted FFUPLINK has manually to be reconnected? WAN on Easybox did get IP lease, FFUPLINK not

testing

everloop2, 2018-05-17:
seems to be fixed, see: https://github.com/freifunk-berlin/firmware/issues/561

closed 2018-05-17 by everloop2