freifunk-berlin/firmware #545: notunnel/default: Uplink does not get announced by OLSR

torte71, 2018-04-04:
When flashing the "vpn03"-flavour, this router automatically appears as SmartGW on other routers, which are connected via mesh (and they will use this uplink).

The "default"-flavour does not do this automatically, I have to add "0.0.0.0" to the HNA list, otherwise this router will neither show up as SmartGW on mesh partners, nor will they use this uplink.

SvenRoederer, 2018-04-04:
sounds to me, that the "uplink has internet" test is failing and for this the smartgateway-capability will not be announced.

please look at 
* https://github.com/freifunk-berlin/firmware-packages/blob/v1.0.0/defaults/freifunk-berlin-olsrd-defaults/uci-defaults/freifunk-berlin-olsrd-defaults#L41
* https://github.com/freifunk-berlin/firmware-packages/blob/v1.0.0/utils/luci-app-ffwizard-berlin/luasrc/tools/freifunk/assistent/olsr.lua#L65

torte71, 2018-04-05:
Weird, today it works as expected. But thanks for pointing out, where to look, if this happens again.
Should we wait some time before closing it? I can't think of anything that I could have misconfigured, like wrong/outdated vpn keys, etc.: Mesh definitely worked, as I was testing 802.11s and had an eye on OLSR neighbours and Inet access was the first thing I've checked.

SvenRoederer, 2018-04-07:
> On router: ping -c 1 -q -I ffuplink 194.150.168.168 # same for 85.214.20.141 and 213.73.91.35, see dyngw plugin: Times out

* can you tcpdump the packages on interface ffuplink?
* do these packages arrive at the uplink-router (fritzbox)
* what is ``` ip route get x.x.x.x oif ffuplink``` resulting

torte71, 2018-04-07:
* can you tcpdump the packages on interface ffuplink?
  - `tcpdump -nli ffuplink host 194.150.168.168` to reduce output (lot of ARP requests from other hosts inside my 192.168.2.0 network)
  - an `ARP, Request who-has 194.150.168.168 tell 192.168.2.69, length 28` for each ping, **but no `ARP Reply`**
**Doesn't look good!**

* do these packages arrive at the uplink-router (fritzbox)
  - On the Fritzbox: `tcpdump -nli lan proto ICMP and host 192.168.2.69` (192.168.2.69 is ffuplink)
    - does not show any activity, when pinging 194.150.168.168 from CPE210
    - does show ICMP packets, when using filter "host 192.168.2.61" (wan) and pinging from CPE210 via wan interface (so the filter is ok)

* what is ip route get x.x.x.x oif ffuplink resulting
```
root@torte-cpe210-1:~# ip route get 8.8.8.8 oif ffuplink
8.8.8.8 dev ffuplink  src 192.168.2.69
    cache
root@torte-cpe210-1:~# ip route get 194.150.168.168 oif ffuplink
194.150.168.168 dev ffuplink  src 192.168.2.69
    cache
```

torte71, 2018-04-07:
Why an "ARP Request" for 194.150.168.168 at all?
Shouldn't that rather be an ICMP packet to 194.150.168.168 with the MAC-adress of my Fritzbox as destination? 194.x.y.z is defnitely not on my lan.

bobster-galore, 2018-04-07:
Just wanted to check on my device, there a different interesting things:
``` 
Sat Apr  7 20:32:27 2018 daemon.notice netifd: ffuplink (1050): udhcpc: sending renew
Sat Apr  7 20:32:27 2018 daemon.notice netifd: ffuplink (1050): udhcpc: lease of 10.230.192.218 obtained, lease time 300

Sat Apr  7 20:32:45 2018 daemon.notice netifd: wan (1045): udhcpc: sending renew
Sat Apr  7 20:32:45 2018 daemon.notice netifd: wan (1045): udhcpc: lease of 10.230.192.238 obtained, lease time 300
``` 
ping from pc to both addresses is ok.

torte71, 2018-04-08:
Now I have this effect on a second CPE210 as well. And it seems reproducible.
* After each reboot, there is no connectivity to the internet via ffuplink, only my local network is reachable (checked with "ping -I ffuplink ...".
* When doing "/etc/init.d/network restart" afterwards, ffuplink connectivity to internet starts  working.

(both checked several times)

I see a different order when the WAN interface is detected: After a reboot, it gets detected by the hotplug scripts **after** ffuplink, whereas after a "network restart" it gets detected **before** ffuplink. 

After reboot:
```
root@torte-cpe210-2:/# logread |grep ff-userlog
Sun Apr  8 11:29:16 2018 user.notice ff-userlog: ffuplink connection will be established
Sun Apr  8 11:29:17 2018 user.notice ff-userlog: ffuplink connection has been established
Sun Apr  8 11:29:17 2018 user.notice ff-userlog: ffuplink interface is up
Sun Apr  8 11:29:17 2018 user.notice ff-userlog: creating ffuplink ip-rules
Sun Apr  8 11:29:17 2018 user.notice ff-userlog: ffuplink-interface is setup
Sun Apr  8 11:29:20 2018 user.notice ff-userlog: WAN interface is up
```

During /etc/init.d/network restart:
```
root@torte-cpe210-2:/# logread -f |grep ff-userlog
Sun Apr  8 11:31:32 2018 user.notice ff-userlog: WAN interface is up
Sun Apr  8 11:31:34 2018 user.notice ff-userlog: ffuplink connection will be established
Sun Apr  8 11:31:34 2018 user.notice ff-userlog: ffuplink connection has been established
Sun Apr  8 11:31:34 2018 user.notice ff-userlog: ffuplink interface is up
Sun Apr  8 11:31:34 2018 user.notice ff-userlog: creating ffuplink ip-rules
Sun Apr  8 11:31:35 2018 user.notice ff-userlog: ffuplink-interface is setup
```

torte71, 2018-04-08:
I diff'ed the sorted output from `ip route show table all` directly after reboot and after "network restart". The routes to my Fritzbox (192.168.2.1) and "default" use the "br-wan" interface after a reboot, but "ffuplink" after "network restart":
```
diff -u -U 0 rules-after-reboot.txt rules-after-network-restart.txt
--- rules-after-reboot.txt	Sun Apr 08 12:12:28 2018
+++ rules-after-network-restart.txt	Sun Apr 08 12:12:18 2018
@@ -1 +1 @@
-- after reboot -
+- after network restart -
@@ -7,2 +7,2 @@
-192.168.2.0/24 dev ffuplink  proto kernel  scope link  src 192.168.2.82
-192.168.2.1 dev br-wan  proto static  scope link  src 192.168.2.72
+192.168.2.0/24 dev ffuplink  proto kernel  scope link  src 192.168.2.83
+192.168.2.1 dev ffuplink  proto static  scope link  src 192.168.2.83
@@ -14 +14 @@
-broadcast 192.168.2.0 dev ffuplink  table local  proto kernel  scope link  src 192.168.2.82
+broadcast 192.168.2.0 dev ffuplink  table local  proto kernel  scope link  src 192.168.2.83
@@ -16,2 +16,2 @@
-broadcast 192.168.2.255 dev ffuplink  table local  proto kernel  scope link  src 192.168.2.82
-default via 192.168.2.1 dev br-wan  proto static  src 192.168.2.72
+broadcast 192.168.2.255 dev ffuplink  table local  proto kernel  scope link  src 192.168.2.83
+default via 192.168.2.1 dev ffuplink  proto static  src 192.168.2.83
@@ -37 +37 @@
-local 192.168.2.82 dev ffuplink  table local  proto kernel  scope host  src 192.168.2.82
+local 192.168.2.83 dev ffuplink  table local  proto kernel  scope host  src 192.168.2.83
@@ -45 +44,0 @@
-local fe80:: dev lo  table local  proto none  metric 0  pref medium
@@ -50 +48,0 @@
-local fe80::c001:3ff:fe90:c8cb dev lo  table local  proto none  metric 0  pref medium
@@ -51,0 +50 @@
+local fe80::e8d6:4fff:fec5:c383 dev lo  table local  proto none  metric 0  pref medium
```

pmelange, 2018-05-11:
Please try giving the ffuplink interface a metric of 1, like suggested in #560.

torte71, 2018-05-11:
OK, I will check this the next days.
IIRC, the problem was, that there was always only one "default" route set, instead of adding it for each interface : The last interface, that received a dhcp-lease simply overwrote the former default route. So after a reboot, when the WAN port came up later, the ffuplink didn't know how to reach outside addresses anymore. But after a "network restart", ffuplink came up later, so access to the outside was possible for this interface.
If the different metric will lead to two separate default routes instead of just one, then your solution should indeed fix this issue.

torte71, 2018-05-13:
@pmelange - that works! With ``option metric '1'`` in the section ``config interface 'ffuplink'`` from ``/etc/config/network`` there are now two default routes created, one for br-wan and one for ffuplink. And the SmartGW gets announced reliably. So: 👍 for https://github.com/freifunk-berlin/firmware/issues/560

SvenRoederer, 2018-05-13:
I had once seen this also on my router. I fixed it by https://github.com/freifunk-berlin/firmware-packages/commit/668d294d9dd2277a0abbb0b7b6c9284779fc8c8b  probably it might also be good to not request a "default gateway" for the ffuplink interface.

torte71, 2018-05-13:
@pmelange - yes, that alternative works here, too (setting ip4table and ip6table)

SvenRoederer, 2018-05-13:
Let's discuss this further in the common issue #561.

closed 2018-05-13 by SvenRoederer