freifunk-berlin/firmware #547: ffuplink should keep its MAC address

torte71, 2018-04-08:
Every reboot (or rather `/etc/init.d/network restart`), the ffuplink interface chooses a different MAC-address. This can eat up the leases in the local network and make difficulties e.g. if you have a MAC-adress filter on your net.
A fixed address can be set by adding an `option macaddres 'aa:bb:cc:dd:ee:ff'` to the ffuplink_dev device config.

Ideas:
* If the "macadress" option is missing, create an address and add the option to the interface configuration, so it gets remembered
* Option a) Derive the address from the eth0 interface, e.g. a fixed part like "d2:ba:be" and the three last bytes of eth0's macadress. This would stay constant even after a "reset to firmware defaults".
* Option b) Let veth choose the complete address and then remember it. Depending on the logic of the veth module, this may reduce the risk of duplicate addresses. But it will not stay constant over firmware resets

SvenRoederer, 2018-04-08:
what about https://de.wikipedia.org/wiki/Locally_Administered_Address in contast of stealing foreign adresses

torte71, 2018-04-08:
This is exactly, what the "2" in the starting "d2" means:
Bit#1:set => local OUI (not universal)
(and Bit#2:clear => unicast)
https://www.itwissen.info/MAC-Adresse-MAC-address.html

Same for the veth kernel module: It creates the address using eth_hw_addr_random() that calls eth_random_addr() which takes care of the bits
https://github.com/torvalds/linux/blob/master/drivers/net/veth.c#L408
https://elixir.bootlin.com/linux/v4.3/source/include/linux/etherdevice.h#L258
https://elixir.bootlin.com/linux/v4.3/source/include/linux/etherdevice.h#L218

torte71, 2018-04-08:
But sure, the fixed part in a) is not necessary, this was just inspired by our AdHoc adresses. Using the complete eth0 MAC adress and only setting the "local OUI" bit has the highest chance of being unique (unless it already had or gets that bit set, see (*)). Our busybox "ash" allows bit operations, so this is doable in a script.

(*) I remember I once had a wifi-repeater that manipulated the first byte of the MAC-address of its clients (probably in the same way, i.e. set "local OUI"). Not sure, how it would behave, if that bit was already set, but if it caused problems, this could be a cause for connectivity problems of the ffuplink interface, like here: https://github.com/freifunk-berlin/firmware/issues/532

pmelange, 2018-05-10:
I am working on making a new bbbvpn which uses tunneldigger.  While creating a uci script to set it up, I needed the following code snippit.  Maybe it helps?
```
#!/bin/sh

# start the first two bytes with "bb:bf" for Berlier BackBone Freifunk
MAC="bb:bf"
for byte in 3 4 5 6 7 8
do
        MAC=$MAC`dd if=/dev/urandom bs=1 count=1 status=none | hexdump -e '1/1 ":%02x"'`
done
echo $MAC
```

torte71, 2018-05-11:
@pmelange
* Why 8 byte for a MAC address instead of 6?
* Starting with "B" means a unicast address. Is that intended? (See above link about the meaning of the i/g and u/l bit)

pmelange, 2018-05-11:
Oops, I was only playing around with code to make a UUID, and quickly renamed it to work for a MAC for this thread.  6 bytes is correct.  At the same time, I also want to have a static MAC for the bbbdigger interface (dhcp-client reasons)

Unicast, in this case, would be correct.  There wouldn't be any reason to use multicast.  But in the long run, since it is encapsulated in a l2tp tunnel, the only devices which would ever see the MAC is the tunneldigger-broker and the client.

pmelange, 2018-05-11:
Ok, right.  Unicast and not multicast.  But bb:bf sounds so nice....  Anyhow, it's not so important how the MAC starts, as long as it's unique.  

But as a code snippit, it's at least a start to solve this issue.

pmelange, 2018-05-14:
I would really like to have the ffuplink keep it's mac addr across reboots.  I think this is a very important feature for those who set up advanced routing at the next hop.  Especially those who assign addresses to a "guest network" (think fritz box), would like to have all freifunk traffic go through a known IP address (think additional firewalling or statistics), or have their own tunnel with a fast(er) pc or raspi for freifunk traffic.

pmelange, 2018-05-14:
@torte71 I decided to go with a mac addr which starts with b6 (Berlin 6ackbone).  Maybe for the ffuplink devices, we could use something like fe (Freifunk Eplink) or fa:dd (Freifunk ADDr).

SvenRoederer, 2018-07-25:
fixed in https://github.com/freifunk-berlin/firmware-packages/commit/355428c00f47b81f4a534b61087299ac8ece3c3c

closed 2018-07-25 by SvenRoederer