freifunk-berlin/firmware #548: alternative traffic-shaping script

SvenRoederer, 2018-04-10:
I just found this in Gluon: https://github.com/freifunk-gluon/packages/tree/master/net/simple-tc

Do you think it's a lightweight alternative to the current qos script?

bobster-galore, 2018-04-10:
Since there is no documentation about the package it's hard to guess.
There has to be some kind of traffic control otherwise the connection will be synchronous to the lowest common speed. Basically it's fast forward for ack-packages and avoiding congestion, the rest is just stuff for special purposes like telephony, stream and others. tc contains all sorts of classes, discs and algorithms, which is nice for academic research but probably not necessary to do a functioning traffic control.
Having been written or introduced by @NeoRaider and in use on a broader scale at gluon I think we should give it a try. How do u know if it is lightweight, from name?

torte71, 2018-04-11:
@bobster-galore - you know it's lightweight by looking at the source: Just a single, 300-line file: https://github.com/freifunk-gluon/packages/blob/master/net/simple-tc/src/simple-tc.c - and it depends (apart from the default libs) only on libnl-tiny, that we use as well.

It only sets input/output rate, no special handling for ACK, congestion, etc.. To see, how that behaves in real life, it might be quicker to run tests, than digging into kernel packet scheduler (do bulk bidirectional data transfer on a single socket, as well as on separate sockets, together with a slow client on a different connection). Or ask someone, who has already done that.

That package looks very interesting IMO.

SvenRoederer, 2018-05-02:
package added to SAm0815_experimental builds

bobster-galore, 2018-05-17:
How can we test it?
I can imagine to put it on a router and do a simple test, just downloading a big file while uploading a big file. When both operate on the given bandwidth it is nice otherwise if it is slowing down download speed to upload speed it is not.
What do u think?

SvenRoederer, 2018-05-18:
This might work, just give it a try and then we will see.

bobster-galore, 2018-06-09:
Ok, did a quick test with  Freifunk Berlin 2018-alpha 6f46d18 on cpe210 v1.1
First I was downloading recent Ubuntu desktop and uploading Ubuntu desktop to the router:/dev/null after the test router. up- and download where quite different.
I put the alpha to the cpe.
While downloading recent Ubuntu desktop I send a 20MB email since I could not reach the other router (probably the first test was not concurrent on the same way since the connection was thru wireless and olsr).
I watched traffic by vnstat -l -i <interface> which is the live (-l) view on an interface. So all traffic was going thru the test router only interface.
The download speed was significant over upload speed, I think the small tc package is sufficient for our needs. (SAm0815_experimental olsr is not working)
Shall I check smthg else or can I revert to hedy 101?

bobster-galore, 2018-06-11:
I'm somehow irritated, did u do anything else beside including it to the build?
E.g. default settings or some findoutandsetitrihgt-magic?
Or is it only living in the system? If this is the case I did the wrong test.

SvenRoederer, 2018-06-11:
the package was just enabled to be build, no configuration was done at all.
I think /etc/config/simple-tc needs to be configured

bobster-galore, 2018-06-12:
So far I can say that my previous test didn't test simple-tc.
Now I installed it by opkg update & opkg install simple-tc.
I configured /etc/config/simple-tc like that:
< 
# Example config

config interface 'ffuplink'
       option enabled '1'
       option ifname 'ffuplink'
       option limit_egress '1000' # 1000 Kbit/s
       option limit_ingress '5000' # 5000 Kbit/s
>
Then I set /etc/config/qos enabled=0.
After restart qos is still in autostart, simple-tc is not and there is no used ffuplink in that experimental notunnel.

simple-tc ffuplink 5000 1000 gives
invalid interface: ffuplink
surprisingly it's not called at startup.

To me it seems that there is more to be done 4 using it. My first thoughts concerned only luci-stuff and variables in uci, think it's more.