freifunk-berlin/firmware #549: Change Smartgateway behavior to respect Bandwidth

nExXxuS, 2018-04-12:
At the moment the smartgateway configuration doesnt respect the capacity for selection, because 'SmartGatewayDividerEtx' is unset, which renders the SmartGW feature and therefore freifunk unusable in some circumstances where you have smart gateways with almost similar ETX.

Reference: http://olsr.org/git/?p=olsrd.git;a=blob;f=README-Olsr-Extensions;hb=HEAD

nExXxuS, 2018-04-13:
Furthermore we should consider lowering the SmartGatewayTreshold, in order to avoid sticking to slower gateways, if a new comes up.

bobster-galore, 2018-05-17:
As the berlin community seeks to leave ancient olsr-versions, I see no reason to think about funny old olsr settings.
If we stick to olsr, we can think about it if it's still available in recent versions (0.9.6.2+) or olsrv2. We learned that olsr SmartGateway is bad with or without.

booo, 2018-08-21:
@nExXxuS could you create a patch for your proposal? Maybe split the proposed changes in two commits. One for SmartGatewayDividerEtx and one for SmartGatewayTreshold.